bancika/diy-layout-creator #273: Can't open DIYLC 3.28.0 OSX  on 10.7.5 - Java version 8 update 71

ilcaccillo, 2016-01-26:
Hello,
I unzip diylc-3.28.0-osx.zip and then click on the "DIYLC" file and it doesn't open, nothing happens at all.
Right click and Open also has the same result.

I already had the setting "Allow Applications Downloaded From" set for "Anywhere"

I'm able to open DIYLC version 2

I'm able to open DIYLC 3.28.0 (multi-platform) although I have to use the "diylc.far" file,  I dont know how to use or open the run.sh

The problem happens also with other different V3 releases for MAC

I'm running OSX 10.7.5 and have Java 8 update 71 instaled

Thanks
Best regards

bancika, 2017-04-24:
Is this still an issue with 3.38?
Thanks and sorry for not responding earlier.

Modbuss, 2017-04-29:
Yes it is still an issue with 3.38, just tried downloading and installing and get a message saying "DIYLC is damaged and cant be opened and should be moved to the trash."

bancika, 2017-05-01:
Did you enable 3rd party apps. Ironically, OSX is the one that damages the app by disabling it and then it complains about it :)

I have verified that app works fine under OSX if java 6 is present and 3rd party apps enabled

Modbuss, 2017-05-01:
Yep I got it to work now, pretty dumb on OSX's part if you ask me. Works great now! Thanks!!

bancika, 2017-05-03:
thanks, closing

closed 2017-05-03 by bancika