bancika/diy-layout-creator #276: App locks while trying to delete

RightBrainAmp, 2016-05-04:
I've run into a major issue that keeps occurring. When I try to delete a component I generally click to highlight the component, then right click to bring up the menu, then select delete or cut depending on whether I still need the component. A majority of the time, when I right click to bring up the menus, it will simultaneously bring up both the menus and the properties box at the same time. If this happens, I'm not able to select a menu option until I close the properties box for that component. Recently, the app has started to hang when this happens not allowing me to close the property box, select any menu items or exit out to try again. I'm also not able to do any keyboard shortcuts at this point so I am frozen in that quagmire and have to force quit. Luckily the autosave function keep up pretty close to real-time so I don't loose much data.
I will admit that part of the problem is my Magic Mouse is not the best tool for doing this. For me in general, working on drawing, design, and CAD, when I want to delete, cut, copy, and/or past a component or object, I select that object then use a keyboard shortcut to perform the operation. Unfortunately DIYLC forces me to right click to bring up the menu. I am a Mac user since the first Mac in '84 and this whole Right click on the mouse thing is not part of our legacy! It's an old IBM thing! 
DIYLC v3.22.0
[image]

bancika, 2016-12-15:
interesting. Not sure how macs work, never had one. So right click opens property editor? It's supposed to open either on double-click or from the context menu.

bancika, 2019-01-12:
this should be fixed in one of the recent versions

closed 2019-01-12 by bancika