bancika/diy-layout-creator #278: Vertical dimension not .1 inch

N9XYP, 2016-11-25:
I was trying to make a Radio Shack protoboard and was have a lot of problems. A minor one was the scale slider. The lack of numbers make adjustments hard. The main problem was the vertical dimension is not the same scale as the horizontal. I made the scan of the board, using Photoshop to make sure the holes were .1" and the ten holes were 1". If the image was scaled to fit .1" on the y-axis the x-axis was off and the reverse. Saving out the image and measuring in Photoshop, the 10 lines measure .9" not 1".

Would it be possible to open the image and scale it based on the internal dpi?

pcd.diy.jpg is not a picture. I can't attach a diy file or a zip. It says I can add a zip but it won't let me.

[image]
[image]

bancika, 2016-12-21:
not sure I understand the issue. Is the photo itself scaled 9x10 or DIYLC scaling does it badly? The later is unlikely. I tried to import a perfectly square image, scaled it up and down and it remained perfectly square in all cases.

I have added numerical box next to the slider so you can type in exact scale. This will be included in the next release.

N9XYP, 2016-12-21:
The picture above is exported from DIYLC. The columns are .1" apart but the rows are .09" apart. The scaling was done in DIYLC to make it fit the spacing and the snap to grid was turned on to make sure my lines were spaced correctly. The jpg file is the DIYLC project file.

bancika, 2016-12-27:
Tnx. I still don't understand how DIYLC should've handled this case better. The JPEG of the board is obviously not perfectly proportional. I tried to crop the center part of the photo, exactly 7x7 cells large and it's 208x199px, i.e. not square (see below). So no matter how you scale it in DIYLC, it will remain non-proportional. The only thing we can do is have separate horizontal and vertical scale controls so you can tweak the two separately. Or try to fix the proportions in paint before importing the image.

[image]

N9XYP, 2016-12-27:
It is a moot point now. Just restarted the project after receiving notice of your reply. Everything is working fine in the latest version. Thanks for investigating.

closed 2016-12-27 by N9XYP