bancika/diy-layout-creator #282: blank saved projects

mbwhite81, 2016-12-31:
Hi,

version 3.34 on Mac OsX 10.12.
I work on a project, and save it. When I re-open the file the project is blank.

Thanks

bancika, 2017-01-04:
can you please upload the app log from logs folder from the time that happened?

mbwhite81, 2017-01-04:
hi, I've attached the zip file to this message

best

[log.zip](https://github.com/bancika/diy-layout-creator/files/685405/log.zip)

bancika, 2017-01-06:
nothing suspicious in the logs, can you please make a diy file that opens empty and send that to me so I can analyze it? Maybe it has something that prevents DIYLC from reading it.

mbwhite81, 2017-01-06:
I have one that I've tried again to save, just 20 minutes ago.
I'm pasting the logs as well.
[log.zip](https://github.com/bancika/diy-layout-creator/files/690376/log.zip)
[cj11_layout.diy.zip](https://github.com/bancika/diy-layout-creator/files/690377/cj11_layout.diy.zip)

mbwhite81, 2017-01-09:
I've just downloaded the new version 35.0.
Same issue

bancika, 2017-01-11:
it opens with one eyelet board on the top of the drawing. There's nothing shown when it opens, but when I scroll up there's a board. Is that what you drew?

mbwhite81, 2017-01-16:
I don't remember. Maybe. I've tried so many times that yes, it could be right.
So, is it an issue with my mac?

bancika, 2017-01-16:
Hmm, if that's what you drew, I don't see an issue. There's one component and it opens properly. When you open the file can you scroll up to see the component? Try clicking the tiny button in the bottom-right corner, it will show a thumbnail of the whole project and you can see what's there are scroll easily.

mbwhite81, 2017-01-17:
yes, I zoomed out the entire project, but there's nothing in it.

bancika, 2017-05-06:
I managed to get a mac (sierra) to try this, but your project file opens nicely. Here's what I see when it opens (after scrolling all the way to the top)
[image]

closed 2018-12-22 by bancika