bancika/diy-layout-creator #288: Requires legacy version of Java on OSX

chipshorter, 2017-01-26:
On OSX Sierra, the message 'To open "DIYLC" you need to install the legacy Java SE 6 runtime'.

Although I'm sure this will work, because I did this for V3.28 of DIYLC on a previous MAC, I'm reluctant to install such an old version of Java now because of security issues etc.

Is it possible to allow a recent version of Java to be utilised?

For info, the latest Java version on OSX is:

java version "1.8.0_111"
Java(TM) SE Runtime Environment (build 1.8.0_111-b14)
Java HotSpot(TM) 64-Bit Server VM (build 25.111-b14, mixed mode)

bancika, 2017-01-26:
I must start with - f'ing Mac :)

The reason I kept DIYLC on an old java version in the first place was that
few years ago Apple was miles behind on Java versions compared to win/linux
and people were complaining that they couldn't run if I compiled with Java
1.7.

For the next release I'll make 1.6 and 1.8 builds separately. Keep an eye
on v3.36

Cheers,
Bane

bancika, 2017-01-26:
I pushed a new version out sooner, give it a try.

https://github.com/bancika/diy-layout-creator/releases/tag/v3.36.0

chipshorter, 2017-01-26:
Wow, I wasn't expecting a response so quick! Thanks, but not all good news I'm afraid.

I downloaded the 1.8 version you just made but it still gives the same message as before. I'm an experienced java developer myself, but I do my development on Windows, whereas the MAC is my "hobby" computer, so I'm not really sure why this occurs.

Let me have a dig around and see if I can find some reason why it's insisting I install 1.6, It seems only fair that I put a bit more effort into this myself, then maybe I can get back to you with the results of my findings.

bancika, 2017-01-26:
it could be due to the way the app is packaged. Looks like jarbundler has
been decomissioned in newer OSX and replaced by javapackager. I'll see how
complicated is to change the deployment code to use that.

chipshorter, 2017-01-26:
I did find something on stack overflow on the thread linked below which looked promising  

http://stackoverflow.com/questions/19563766/eclipse-kepler-for-os-x-mavericks-request-java-se-6/19594116#19594116

The 1.6 message has now gone away, but it's been replaced with '"DIYLC" is damaged and can't be opened. ...'

If you don't have any luck with the packager, I can always run the multi-platform version I guess, but it would be nice if we can find a way to get this to work as it should. I'll keep digging.

chipshorter, 2017-01-26:
OK, some further progress...

I found another link which expanded on the previous one...
https://code.google.com/p/android/issues/detail?id=65730#c3

I needed to create some directories and symbolic links, but now I can run OK from a command line (which I couldn't previously) by running...

> DIYLC.app/Contents/MacOS/JavaApplicationStub

However I still get the '"DIYLC" is damaged ...' when I try to launch from the icon, so it's not a perfect solution by any means.

The fonts look a bit blurred, but that's another issue.

bancika, 2017-01-27:
As I said, f'ing Mac :) Always some complications with it.

If you use eclipse and have some spare time, maybe you could try to rebuild
the app for new java using new bundler. All the source is check it to git
hub. In diylc-swing is the build.xml that has task for OSX build, along
with other things. It uses jarbundler from ant-lib folder. As far as I
understand, we need to replace the call with the new bundler. Here's some
info
http://dclunie.blogspot.rs/2014/10/keeping-up-with-mac-java-bundling-into.html

I can't do much at the moment because a) I don't have a mac to test on and
b) the workaround is very simple - just install java 6, so the issue is not
a show stopper.

Cheers,
Bane

chipshorter, 2017-01-27:
I'll have a go at building as you suggest as soon as I get chance and let you know. I wasn't expecting you to put a lot of time into this, and as you said it's not a showstopper.
Thanks
Chris

bancika, 2019-01-12:
merging with #370

closed 2019-01-12 by bancika