bancika/diy-layout-creator #290: Breadboard power strips not aligned with grid

fuser-invent, 2017-04-06:
I'm running the newest version 36.0 and noticed that the power strips on each side of the breadboard don't align with the grid. Instead of 5 connection spots per grouped rail there are 4. There are also 6 groups per side instead of the 5 you would see on a physical breadboard.

bancika, 2017-04-07:
As far as the alignment goes, I made it based on this breadboard https://cdn-learn.adafruit.com/assets/assets/000/002/602/medium800/learn_arduino_breadboard_half.jpg?1396784960
It also has shifted power strips for 0.05". I could try to align them, it would make it easier to use but less true to the original.

will fix the rest.

Cheers

bancika, 2017-04-07:
fixed for 3.37

closed 2017-04-07 by bancika

fuser-invent, 2017-04-07:
Thanks Bancika. I realized I may not have explained it well and made an image to show what I mean. http://imgur.com/a/PPvDK --- I also wanted to say that this software is awesome and was able to help more with it.

The image I made shows the way a physical breadboard groups the power strip inputs and the way they are grouped in the software. The number of inputs on the strips in the software aren't the same as a real breadboard. The alignment when making connections is also off if "snap to grid" is enabled.

Even if the breadboard is accurate to real life the jumpers don't align to real life because the center of the inputs on the middle of the board are on the grid lines like everything else in your program but the center of the power inputs are offset by 1/2 a grid space so they are inside a grid square and not on the grid lines.

Looking at the way components are made, I wouldn't know how to automatically bypass the normal "snap to grid" function and "snap to component connection point" instead. It seems like it would be easier to shift the power inputs up or down by 1/2 a grid space so they all align. I know it will be less true to the original but think it will make more sense visually in the software.

Thanks for everything you do.

bancika, 2017-04-08:
I understood what you meant the first time. 

For the next version I updated breadboard component with 3 new settings that you can change from the dbl click dialog - power strip alignment (offset like original or inline which is easier to use), board size (half with 30 rows of full with 63) and orientation so you can rotate the board as needed. I also changed the way power supply grouping works, so we have 5 inputs per group instead of 4.

Cheers

bancika, 2017-04-10:
Please try v3.37 I just released

fuser-invent, 2017-04-10:
Totally perfect, thanks!