bancika/diy-layout-creator #292: Corrupted file on OS X Yosemite - Unable to run v38.0

batchas, 2017-04-22:
I tried to open DIYLC Gen3 v38.0 on OS X Yosemite. It says to install an older JAVA version. I did that.
Restarted. Tried to open DIYLC and an error message appears: "The file is corrupted. It is recommended to move it to the Trash" :(
Downloaded again from here, unpacked, tried again. No success. Same error message.

bancika, 2017-04-22:
thanks for reporting. Does v37.0 work for you?

batchas, 2017-04-22:
There's no version which works on my macbook pro with Yosemite.

I saw that others have the same issue: https://groups.google.com/forum/#!topic/diy-layout-creator/VWZ5m5Dir6I

bancika, 2017-04-22:
did you enable 3rd party apps like shown here http://it.nmu.edu/docs/allowing-third-party-applications-install-macbook

batchas, 2017-04-23:
I always get another kind of message in this case, so I did not think about that. But it solved it. Thanx for helping!
SOLVED :)

bancika, 2017-04-23:
Great. If I had a dollar every time this happened, I would've collected enough to have DIYLC listed in the official apple store :)

closed 2017-04-23 by bancika