bancika/diy-layout-creator #294: DIY Layout Creator Not Started on Win7 32bit

bvvuz, 2017-05-03:
DIY Layout Creator (v3.39.0) not Started on Win7 32bit (admin privileg ON), old version (Pascal Revision) worked (new version normal worked Win2003, Win7 not worked),
And, viewing (test on Win2003), looking attach screenshots.

[image]
[image]
[image]

bancika, 2017-05-03:
This is expected. That kind of jack component is not implemented in v3, but
there are few other jack components that you can use instead

bvvuz, 2017-05-03:
Thank you, but still, how to run the program on Win7 ?

bancika, 2017-05-03:
But it runs on your screenshot? I'm not sure what I'm looking at then.
How does "not run" manifest?

bvvuz, 2017-05-03:
These screenshots are made in Windows Server 2003 (on this platform both programs work), on Windows 7 32bit the new version of the program does not start at all :(...

P.S. After about one hour I will be able to make a copy of the record from Windows 7 (event log) that appears when the program starts...

bvvuz, 2017-05-03:
Sorry me, please, in Windows 7 so far it was not possible to create a log (for some reason the journal is empty), I'll try it in the evening, maybe a log entry for this event will appear in the log (the log when the new version of the program is launched).

bancika, 2017-05-03:
is anything written in diylc log located at c:\Users\<your name>\.diylc\logs

bvvuz, 2017-05-03:
Oh, sorry, I did not think about it, in the evening I will try to attach this log.

bancika, 2017-05-03:
there may not be the log there if the app failed to start completely. In
that case it's java or windows issue. But if the app started and crashed it
should say something in the log

bvvuz, 2017-05-03:
Ok, thanks, try to check it in the evening :)

bvvuz, 2017-05-03:
I'm sorry, but i do not have anything at the specified address either :(. By the way, my operating system is Russian (Win7 ru 32bit), all the paths (for logs) you specify in my Windows (Win7) are also in Russian ?

This problem only with JAVA applications, other programs (not JAVA) all work fine (as an example the old version of DIY Layout Creator, not JAVA?). By the way (for comparison), the old version of DIY Layout Creator (from the first screenshot) on what path creates (if creates) similar logs?

bancika, 2017-05-03:
the old version of DIYLC is not java-based and it doesn't produce any logs. Can you try to double click on diylc.jar instead of exe? Does that work?

bancika, 2017-05-03:
Another thing to try is to open command prompt (in admin mode), go to DIYLC folder and type in 
'''
java -Xms512m -Xmx2048m -Dorg.diylc.scriptRun=true -Dfile.encoding=UTF-8 -cp diylc.jar:lib org.diylc.DIYLCStarter
'''

if there's any error you'll see it in the console

bvvuz, 2017-05-03:
wow, it's worked, (double click for diylc.jar), opened-starting this programms (just yet there was a message from the screenshot below)

[image]

and (working prog ramm window)

[image]

bancika, 2017-05-03:
if that works, just make diylc.bat file in the same folder and copy this command in there
```
java  -Xms512m -Xmx2048m -Dorg.diylc.scriptRun=true -Dfile.encoding=UTF-8  -jar diylc.jar
```
you can use bat file to run the app

closed 2017-05-03 by bancika

bvvuz, 2017-05-03:
sorry this method not worked :(, clicking diylc.jar - worked, clicking this diylc.bat - not working :(...

bancika, 2017-05-03:
When you run bat file from the console what error do you get?

bvvuz, 2017-05-03:
this message:

C:\Program Files\DIY Layout Creator v3.39.0>Java -Xms512m -Xmx2048m -Dorg.diylc.
scriptRun = true -Dfile.encoding = UTF-8 -jar diylc.jar
Error occurred during initialization of VM
Could not reserve enough space for 2097152KB object heap

bancika, 2017-05-03:
Oh now I get it, it cannot reserve enough memory for DIYLC either because
you don't have enough memory or because it's a 32bit system.
Let's try one more thing, go to diylc.l4j.ini and replace -Xmx2048m
with -Xmx1024m,
that should solve it and you should be able to use exe file to run.

bvvuz, 2017-05-03:
There already was (diylc.l4j.ini):

-Xms512m -Xmx1024m -Dorg.diylc.scriptRun=true -Dfile.encoding=UTF-8

=)

bancika, 2017-05-03:
that's odd, it should let you allocate 1GB of ram even on 32bit windows.
Try with Xmx512m

bvvuz, 2017-05-03:
C:\Program Files\DIY Layout Creator v3.39.0>Java -Xms512m  -Xmx512m -Dorg.diylc.
scriptRun = true -Dfile.encoding = UTF-8 -jar diylc.jar
Error: Could not find or load main class =

bancika, 2017-05-03:
please put the whole command in the same line and do not put blanks around "=" sign.

bvvuz, 2017-05-03:
Yes, thank you very much, victory, it turned out, earned !. This parameters replaced in diylc.l4j.ini for starting only exe (without batch file) ??

bancika, 2017-05-03:
yes, if it works just update l4j.ini file and use exe

bvvuz, 2017-05-04:
Thank you very much for helping me :). 

I just had a question a little off topic. Can I import components from the old version (the first screenshot, from the first message) to the new version (the second screenshot, from the first message)? There, for example, displays the same circuit, but looks different (for example, there are no connectors-jacks) ?

bancika, 2017-05-04:
No, but it imports over 90% of the circuit from the old version, so I think
it's good enough :)

bvvuz, 2017-05-04:
ok, thanks :)

The truth seems to me that for future versions such an option (export-import) would not hurt. I would change the base of components for more convenient navigation of the elements (and later export-import would help in transferring the adjusted database to a new version of the program) :).