bancika/diy-layout-creator #295: component text too close to component graphic

robert-rozee, 2017-05-07:
hi,
    i used diylc quite a bit to create circuit diagrams. these are then exported as PDF and often converted to bitmaps.

the one major problem i experience is that component text (value or name) sits extremely close to the component graphic, usually almost touching. is there any chance that a setting could be added to either allow nudging the text away from the graphic, or to perhaps cause the text to positioned as if it were one or two points bigger?

such a setting could be implemented either on a component-by-component basis, or a global setting in the project settings. i'd be more than happy with a single project setting.

cheers,
rob   :-)

bancika, 2017-05-07:
hi,
do you have a screenshot of an example? This doesn't sound right, text should be centered.

Cheers

robert-rozee, 2017-05-08:
here are a couple of images. the first is a 1200dpi conversion using photoshop: 
[image]

the second is a 3200% magnified view of the pdf, screen captured and cropped:
[image]

the 1200dpi one also shows the grid lines and dots slightly displaced - but this is not a big issue. the 3200% one shows the close distance between text (10 ohm) and graphic for resistor. the lines of trace to the left of and first resistor diagonal also don't quite meet on the left, but again is not an issue.

and here is the pdf generated by diylc:
[E-28 (1C) diagram.pdf](https://github.com/bancika/diy-layout-creator/files/982136/E-28.1C.diagram.pdf)

i've  tried uploading the diy file, but github is determined to not allow me to. tried renaming as xml or txt, yet still rejected, even if in a zip file. am happy to email you the diy file. about the only thing of not about the file is that i do set the grid to 0.1" (ie, imperial).

cheers,
rob   :-)

bancika, 2017-05-10:
I moved the label away from the component. Interesting how it looks in 1200dpi, probably issue with PDF, I don't see any misalignment in the app when zoomed 300% (that's 600px/inch).

robert-rozee, 2017-07-20:
any chance you could generate a new release of the windows version so i can test this?

cheers,
rob   :-)

bancika, 2019-01-12:
Apologies for forgetting, this was released over a year ago.

closed 2019-01-12 by bancika