bancika/diy-layout-creator #296: Cloud access and announcements behind firewall/proxy server do not work

PH0MB, 2017-05-08:
Small minor issue, but I just want to inform.

http://www.diy-fever.com/diylc/api/v1/getCategories.php can be directly accessed using a Internet browser but the DIYLC application log get's a connection refused.  

11:16:27,254  INFO [SwingWorker-pool-1-thread-10] CloudPresenter - Fetching categories
11:16:27,254 DEBUG [SwingWorker-pool-1-thread-10] PhpFlatProxy - Connecting to: http://www.diy-fever.com/diylc/api/v1/getCategories.php
11:16:28,268 ERROR [SwingWorker-pool-1-thread-10] PhpFlatProxy - java.net.ConnectException: Connection refused: connect
11:16:28,314 ERROR [AWT-EventQueue-0] CloudBrowserFrame - Background task execution failed
java.util.concurrent.ExecutionException: org.diylc.plugins.cloud.presenter.CloudException: java.lang.NullPointerException

At home it works OK.

MaBerY

bancika, 2017-05-16:
You'll need to add these params 

-Dhttp.proxyHost=*your proxy* -Dhttp.proxyPort=*your port*

to either diylc.l4j.ini (windows) or run.sh (linux)

PH0MB, 2017-05-23:
Ah! Thanx.
I did not know that.  I thought that configuring the proxy settings in the java control panel was sufficient.
As it turned out it is not. :-(

I tried the recommended settings using v3.40 and it enabled the cloud features and it now checks the update.xml file according to the logs.
The http://www.diy-fever.com/diylc/api/v1/getAnnouncements.php still fails. Also the FAQ link https://github.com/bancika/diy-layout-creator/blob/wiki/FAQ.md issues a 404 http not found error.

closed 2017-05-23 by PH0MB

bancika, 2017-05-23:
> diy-layout-creator/blob/wiki/FAQ.md issues a 404 http not found error.

that's ok, there are no announcements at the moment and I deleted FAQ for
some reason :)