bancika/diy-layout-creator #297: Program says app was not closed normally

GeeTeeCee, 2017-05-08:
Even though I do frequent Saves during editing and always say Yes to save changes upon Exit, when I reopen the file I keep getting a message saying, "It appears that aplication was not closed normally ... "
 
[image]

[Using version 3.40 on XP-Pro]

PS: You might like to fix the spelling of 'application' in that message. :-)

bancika, 2017-05-10:
did it happen only once or every time?

GeeTeeCee, 2017-05-10:
It happened numerous times yesterday when I first installed the application, but doesn't happen today.

Very strange.

bancika, 2017-05-10:
I found a scenario that causes this when running multiple instances of the app at the same time. Was that the case?

GeeTeeCee, 2017-05-10:
Possibly. I was trying out its functionality, so I may have had two instances running without realizing it.

bancika, 2017-05-10:
ok, I fixed that for next version
tnx

closed 2017-05-10 by bancika