bancika/diy-layout-creator #298: Request: ability to lasso objects for grouping

GeeTeeCee, 2017-05-10:
This may be difficult to accomplish in code, but it would be fantastic if, instead of having to click each object separately whilst holding down the CTRL key, we could put a frame around a bunch of objects and then CTRL-G them in order to group them for moving purposes.

bancika, 2017-05-10:
well, you can do it :)

GeeTeeCee, 2017-05-10:
What do you mean?

bancika, 2017-05-10:
you can click and drag the mouse around objects you want to select, then press Ctrl+G and voila!

GeeTeeCee, 2017-05-10:
Okay, when I try that it frames the whole board initially and is very difficult to get the lasso to reduce without moving every object on the board.

bancika, 2017-05-10:
I see. Then I can make right click make the lasso rectangle without moving any components. You'd be able to select stuff inside a large board without moving the board. Sounds good?

GeeTeeCee, 2017-05-10:
I'm thinking it would start as a cross hairs at the point where the click is done, then expand (in both X and Y directions) by dragging until un-clicked.. 

Microsoft Paint's select function is what I have in mind.

bancika, 2017-05-10:
I see what you mean, that's not very easy to do here. Paint is raster-based software and DIYLC is vector, like corel. It works differently. I think that right click to select without moving any components will suffice for this use case.

GeeTeeCee, 2017-05-10:
Whatever you can achieve in the environment will be appreciated. Thanks.

bancika, 2017-05-10:
Ok, this will be included in 3.41

closed 2017-05-10 by bancika