bancika/diy-layout-creator #300: JRE Problem

Sylv1v, 2017-05-13:
Hello (from France),
I am trying to install DIY layout creator. After installation, when it tries to run, a window pops up saying : 
"The registry refers to a nonexistent Java Runtime Environment installation or the runtime is corrupted".
I run Windows 10 64bits built 1607 with the latest JRE package.
Any ideas on what to do to make it work?

Thanks!
Sylvain

bancika, 2017-05-13:
can you try to open this link and verify that java is installed properly

https://www.java.com/en/download/installed.jsp

Sylv1v, 2017-05-13:
Thank you for your prompt answer!
It says : you have the recommended java installed (version 8 update 131) on firefox.
It won't work on Chrome and MS Edge as java is desactivated.

bancika, 2017-05-16:
odd. Is your java 32 bit or 64 bit? What happens if you type "java -version" in the console?

tomlut, 2017-05-17:
Same issue. 32 bit Java (v8 build 131) Windows 10 Pro 64 bit.

bancika, 2017-05-17:
I think you need 64 bit java

tomlut, 2017-05-17:
I'm not using a 64bit web browser so I'm never given that option. Also DIYLC used to work with the 32 bit version.

tomlut, 2017-05-17:
Found the 64 bit version on the manual download page. Installed and DIYLC is now running. Thanks for your assistance.

bancika, 2017-05-17:
Glad it works now. It's something about JRE, doesn't sound like it's DIYLC-related issue. Closing for now.
Cheers

closed 2017-05-17 by bancika