bancika/diy-layout-creator #302: Component to be placed selection on OSX Sierra

aphilip, 2017-05-16:
After placing a component I have the press ESC each time before I can select another component (from the components menu) to place.

bancika, 2017-05-16:
what happens if you don't press escape?

aphilip, 2017-05-16:
The selection stays on the current selected item.

BTW, I have reopened the file and now I do not have the issue again. I will do some further testing to investigate what triggers this behaviour.

bancika, 2017-05-16:
That's odd, haven't seen that. So you click on a component in the toolbox, click on the canvas to place it and it still moves around?

aphilip, 2017-05-16:
Ok, I can reproduce it again. 
I just created a new project. and added a board.
Then when I click eg. resistor the selection stays on Board1. And I can no place the resistor but I can drag the board around.

The problem is not consistent. When I keep repeating the proces:

1. Click component
2. Try to place it.

Sometimes it works, sometimes the previously placed component stays selected.

One other thing to note is that I am running dual screens, with DIYLC on my regular laptop screen. (primary)

Another think I notice is that when I have a few component categories opened in the tree the rendering messes up when I scroll. But that is a different issue.

bancika, 2017-05-16:
can you confirm that you clicked on the canvas when adding the board. You need one click to select what you want to add and another to fix it in place. It shouldn't follow the cursor anymore after the second click.

aphilip, 2017-05-16:
It is not following the cursor. I made a video, I will share you a link. The issue shows around the 2:30 mark

aphilip, 2017-05-16:
https://drive.google.com/open?id=0BwFILELXFvUbV0dJR3d0Q0FsRkE

aphilip, 2017-05-16:
You'll see that it works fine at first, but then at around 2:30 it will start to act up.

bancika, 2017-05-16:
thanks for the video, it helps. Looks like sometimes click on the toolbox doesn't take. You see that after problematic clicks, the text in the status bar doesn't change to reflect the new component that's being created. I'll investigate further
Cheers

aphilip, 2017-05-16:
Glad to be of assistance.

bancika, 2017-05-16:
I found one scenario that could be plausible here. If you click steadily it works, but if your mouse moves even by a pixel between the time you press the mouse button and release the mouse button, it doesn't take. I'll fix for 3.41
Cheers

aphilip, 2017-05-16:
Ok, that was easy to verify. 
You're looking at mouseclick instead of mousedown, and it thinks I'm dragging.
If I click carefully I have no issues, but If I drag just a little with the mouse/finger down it does not work. 

Another side effect of this more sensitive trackpad on the new mbp.

bancika, 2017-05-16:
Yes, I replaced mouse click events with more reliable tree selection events. It works even if you use keyboard to scroll through the list. Put the fix in for the next release.

closed 2017-05-16 by bancika