bancika/diy-layout-creator #310: Lock item with right click

Citysmasher, 2017-05-23:
would it be possible to lock components in place individually by using the right click menu?

bancika, 2017-05-24:
technically it's possible to do. What would be a typical use-case for  this functionality? Can it be covered with the existing functionality to lock an entire layer?
cheers

Citysmasher, 2017-05-24:
Please tell me how I can lock a single item now using layers.

bancika, 2017-05-24:
You cannot, but I'm asking how would you use it. In my workflow I would lock board layer and continue placing components. I never had a need to lock a resistor or a cap. That's why I ask. Maybe what you need can be done differently...or maybe the feature is necessary.

Citysmasher, 2017-05-24:
I would lock components for a number of reasons, but as it stands now you have to lock all of them, which really has no use at all.  You lock components for the same reason you lock the board.  So you can choose items, move items, etc without the board being included.  Same goes for components.

pmfawcett, 2017-05-31:
Personally I really dislike the layer-only paradigm for locking. Usually it's specific things I want to nail down, for a temporary directed purpose.

bancika, 2017-06-01:
This is added to the pipeline for the next release.

sehrgut, 2018-10-04:
I second this: I would love to be able to lock the board most of the time, so layer-level locking would be fine for me.

sehrgut, 2018-10-04:
I actually just came across an instance where it would've been really helpful to be able to lock an IC as I tried to manipulate hookup wires going over it. I would find this useful as well.

EL34xyz, 2018-12-18:
I like this idea
I would like to lock things that are attached to the chassis, like Pots, tube sockets and transformers.

bancika, 2019-01-11:
I started working on this but then hit a wall. It's easy to lock a component, but how do you unlock it :)
The way it works is that locked components don't respond to any user actions...so it's impossible to select them to unlock them.

EL34xyz, 2019-01-11:
> I started working on this but then hit a wall. It's easy to lock a component, but how do you unlock it :)
> The way it works is that locked components don't respond to any user actions...so it's impossible to select them to unlock them.

Maybe right click a list of locked items to unlock it?

bancika, 2020-04-19:
included in 4.5.0

closed 2020-04-19 by bancika