bancika/diy-layout-creator #311: Fill in holes in Marshall style perf board

Citysmasher, 2017-05-23:
Is it possible to fill in holes in the Marshall style board after the design is complete?

bancika, 2017-05-24:
I do not agree with this request. This component represents board like shown below. It comes with the holes. If you were to use it to build an amp, you wouldn't go back and fill in the unused holes after you are done wiring, right?

There's blank board that can be configured to have the same color as marshall-style board and then you can use it to place eyeley/turret components wherever you want. You can use project grid as a guideline where to put them.

[image]

Citysmasher, 2017-05-24:
The reason you use this board other than the look is for the grid to hole spacing.  Especially when designing from scratch.  You need the fixed hole spacing (assigned globally) so you can lay the board out and you are not sure what it looks like until you are done.  It would be nice to simply eliminate the unused holes after before sending the file to CNC board drilling.  Instead you have to choose to not have a fixed hole spacing (unless you do it completely manually using the blank board and then you can't change the spacing globally) or you end up with a bunch of extra holes at the end when you are done.

closed 2019-01-09 by bancika