bancika/diy-layout-creator #314: Importing V2 diy files

NickD9, 2017-05-25:
I have some DIY files created in V2.01 Beta, and they load fine in that version, but the load or import in the latest version errors on trying to pull them in. Any ideas? I can supply one of these older files if it would help.

bancika, 2017-05-25:
sure, please attach some of problematic files and I'll investigate. There's no 100% compatibility between v3 and v2, but most of the project should get imported into v3.

NickD9, 2017-05-25:
[AVR_Test.diy.txt](https://github.com/bancika/diy-layout-creator/files/1030395/AVR_Test.diy.txt)
[IR_Slave.diy.txt](https://github.com/bancika/diy-layout-creator/files/1030396/IR_Slave.diy.txt)

NickD9, 2017-05-25:
Files uploaded. Had to add .txt extension to get them to upload. Tried to zip them but it wouldn't have it...

Nick