bancika/diy-layout-creator #320: Schematic potentiometers reset orientation

blackcorvo, 2017-06-25:
After getting the most recent version of the software, I noticed that after opening a saved schematic, the potentiometers were reset to their default positions, instead of rotated to fit the schematic.
It seems to be a graphic glitch, since the actual connection points are in their expected rotated position, but the symbol for the potentiometer is out of place. They do go back to their correct positions once you move them, though.
I also noticed that the text for the potentiometers doesn't adjust it's position after rotating, like it does for other components.

Citysmasher, 2017-06-29:
I was going to post the same thing.

alixanc, 2017-06-29:
same for me with the schematic jfets on mac os.

blackcorvo, 2018-10-15:
I just discovered a workaround for this issue, sorta.
Instead of rotating the component, use the flip option. It won't re-orientate if you do that.
However, if you NEED to rotate the component, it's going to re-orientate next time you open the file.

Also, I've noticed that if you rotate either 90º or 270º and then flip the component (works the other way around as well), the wiring points get displaced completely outside the component symbol.

There's definitely some issue with the rotate function.

MacFly77, 2019-01-10:
Same issue here DIYLC G3 V47 OSX 10.11.6

bancika, 2019-01-10:
Yes, this was never addressed.

MacFly77, 2019-01-10:
Hope you get to it in the near future ;-)

bancika, 2019-01-10:
oki doc, will include

closed 2019-01-10 by bancika