bancika/diy-layout-creator #321: Transparency Does Not Work

mwelch55, 2017-07-03:
The transparency does not work in the current release (4.40.0).  I tried several versions back to see when it was broken and it is working in version 3.32.0

To make it fail, edit a component such as a resistor, choose the color of the component and select the HSV tab, then slide the control. All it does is change the color to gray and vary the shade of gray.

This is a great tool.  Thanks for creating it.  I will be adding to the donations.

Mike

mwelch55, 2017-09-12:
I found that the transparency work, just not the way I thought it did.  

This is the method I was using that doesn't work any more: edit component, edit body color, click HSV Tab, adjust transparency

The method that works: edit component, adjust "Alpha" control.  

The Alpha control seems to be the new transparency adjustment.

bancika, 2017-09-12:
That was the main method to control transparency all along. I honestly
didn't know about the HSV tab. It shouldn't be used for that purpose :)

closed 2018-12-22 by bancika