bancika/diy-layout-creator #325: High DPI issues

guiweber, 2017-07-14:
The following issues were noted when running the app on a 220 DPI screen (15" at 4K resolution), both Windows (tested under W10) and Linux (tested under Fedora with Gnome Wayland) are affected. I am using the default DPI scaling of 200% under Linux and 250% under Windows.

- The rulers are very small and the numbers are not legible.
- The component menu is not large enough to display all component names and cannot be resized.
- The folder rows in the component menu are not tall enough and the text is truncated. Only the folders are affected, the components themselves seem to have taller rows and look fine.
- The icons are very small, such as the search icon, the 4 icons at the bottom right of the screen, the auto-scroll icon and all other icons in the menus.
- The rows in the BOM windows are not tall enough which makes the text illegible.

[image]
[image]

guiweber, 2017-07-14:
P.S. The About windows is also too small for its content.

[image]

bancika, 2018-12-25:
Found this solution, please check if it works https://superuser.com/questions/988379/how-do-i-run-java-apps-upscaled-on-a-high-dpi-display

> Just found an easy solution on my Windows 10 machine:

> Find java.exe you installed.
> Right click -> Properties
> Go to Compatibility tab
> Check Override high DPI scaling behavior.
> Choose System for Scaling performed by:

rickmccl, 2019-01-02:
Following your suggestion, this seems to be working for me: 

Find C:\Program Files (x86)\DIYLC\diylc.exe
 (same steps as above)
Right click -> Properties
Go to Compatibility tab
Check "Override high DPI scaling behavior" box.
for "Scaling performed by:" choose "System (Enhanced)"

Thanks!

[image]

closed 2019-01-03 by bancika

maeknight, 2023-10-15:
this is still an issue under linux, and the suggested solutions do not work

[image]