bancika/diy-layout-creator #326: install.iss: Source File "<project path>\diylc-swing\config.xml" does not exist error on build

AXington, 2017-08-31:
So, I forked the repo and am trying to add some components, and on the Ant build I'm getting an error in the install.iss file on the Ant build for diylc-swing saying config.xml doesn't exist. Lo and behold, a quick look-see shows it doesn't exist. I don't see anywhere in the project where that's generated, maybe I'm blind, what is supposed to be in that file or how is it supposed to be generated? Once I'm up and running and get my components done, I might take a stab at some of the components requested in https://github.com/bancika/diy-layout-creator/issues/245

bancika, 2017-08-31:
try to start DIYLC from eclipse and it should generate config.xml
not sure why it's required though, I might have to revisit that at some
point.

cheers

AXington, 2017-09-02:
So, I use Jetbrains IDEs (when I use IDEs, typically I'm a VIM guy). And I was having trouble figuring out how to run it in it. So I decided to temporarily use eclipse, and ran into an issue where the newer version of eclipse apparently doesn't support anything below Java 1.7. Anyways, so, just to not keep fuxing around with it, ended up compiling it in IntelliJ and running it in Eclipse. You should seriously consider upgrading to a newer Java. 1.6 and 1.7 are both quite outdated and the older JVMs are riddled with security issues and vulnerabilities. Not exactly a huge issue when dealing with this sort of thing, but it makes it harder for new developers to get up and running. Anyways, PR incoming for adding a 3 way "mustang" style switch.

closed 2018-03-07 by AXington

shouse1953, 2019-01-10:
With Java 8 (191) and MyEclipse 2017, I get:

     [exec] Error on line 17 in D:\code\diylc\diy-layout-creator-3.47.0\diylc\diylc-swing\install.iss: Source file "D:\code\diylc\diy-layout-creator-3.47.0\diylc\diylc-swing\config.xml" does not exist.
     [exec] Compile aborted.

I tried but cannot run diylc from inside eclipse as other xstream parsing errors occur:

5:22:24,957 DEBUG [Thread-2] DIYLCStarter - Finished font pre-loading
Exception in thread "main" java.lang.ExceptionInInitializerError
	at org.diylc.swing.gui.MainFrame.<init>(MainFrame.java:118)
	at org.diylc.DIYLCStarter.main(DIYLCStarter.java:166)
Caused by: com.thoughtworks.xstream.converters.ConversionException: Cannot construct org.diylc.core.Template as it does not have a no-args constructor : Cannot construct org.diylc.core.Template as it does not have a no-args constructor
---- Debugging information ----
message             : Cannot construct org.diylc.core.Template as it does not have a no-args constructor
cause-exception     : com.thoughtworks.xstream.converters.reflection.ObjectAccessException
cause-message       : Cannot construct org.diylc.core.Template as it does not have a no-args constructor
class               : java.util.HashMap
required-type       : org.diylc.core.Template
path                : /map/entry/list/org.diylc.core.Template
-------------------------------

Can you post the missing file please?

bancika, 2019-01-10:
can you try to update? Not sure why it doesn't work for you, but I added a default ctor

shouse1953, 2019-01-10:
I just updated source and to latest version today before I tried to get things into MyEclipse.  

I did end up finding a config.xml from my c:/users/xxxx/.diflc folder, copied it to the source folder and that eliminated the compile issue. 

Now I’m left with just the parsing issue.  Seems it wants a default constructor with no arguments for Template. The last error I posted is the parsing error.

So I added an empty constructor to Template and recompiled. Now I can start the diylc launcher but it still has a few errors. Diylc does start up and the GUI works but the console shows errors while it loads. They reference issues with the multi can capacitor. 

Will you consider moving to a newer java soon? You might want to be on 8 for security reasons. I can’t use your java since myeclispe 2017 won’t allow anything that old. 
 
Don’t get me wrong...I love what you’ve done but can’t contribute if I cant get it working on the newer versions. 

Many thanks,
Steve House 
Www.housekustoms.com

bancika, 2019-01-10:
I would upgrade to java 8 in a heartbeat but MacOS package will stop
working and to build it for Java 8 takes more effort. Still not even sure
it's possible to build Mac apps for Java 8 on a windows machine