bancika/diy-layout-creator #327: OSX tells me latest release is broken :(

hedefalk, 2017-08-31:
[image]

bancika, 2017-08-31:
OSX is idiotic :)

you need to enable 3rd party apps.

closed 2017-08-31 by bancika

hedefalk, 2017-08-31:
Sorry, I totally agree, hate that gatekeeper crap and I don't ever use the app store, but normally I would get a question and I can go in there and override temporarily but persisting for that app but cannot this time, I guess since not native app but Java? 

[image]

Anyway:

sudo spctl --master-disable
restart settings and then:

[image]

Maybe this was in instructions, but I missed it. Thx!

bancika, 2017-08-31:
yup, mentioned on the front page
https://github.com/bancika/diy-layout-creator