bancika/diy-layout-creator #329: Could not open file java.lang.Out of memory error:Java heap space

Swingarm66, 2017-09-18:
my lap top lost power and my 50 meg file wil not open now!!!! Any help would be appreciated many hours into this :(

bancika, 2017-09-18:
doesn't seem like the file is corrupt, more like the app ran out of memory. What OS are you using? 32 or 64bit laptop? We can try to increase memory size.

Swingarm66, 2017-09-18:
Im using win 7 64 bit i dont know can i send you a google drive file? to
see if you can open? I doing a dumble clone and i am going to cry if i have
to start again!! its a laptop

bancika, 2017-09-18:
try to zip it and email to bancika AT gmail DOT com

did you save your changes before laptop lost power? Is that file up to date? if not, you could try with auto-save file that gets saved once a minute (provided you haven't edited any other files after that).

Swingarm66, 2017-09-18:
I'll try to zip, when i open the program again there was no auto save
prompt like there often is

Swingarm66, 2017-09-18:
ok sent

closed 2017-09-20 by bancika