bancika/diy-layout-creator #332: Bypassing Gatekeeper

macumbista, 2017-10-16:
The newer versions of OSX do not allow one to select "apps from anywhere" in the user security preferences GUI. I would like to suggest this method be posted to the FAQ to help new users install the software on OSX:

To get DIYLC to run for the first time on OSX, you need to disable the Mac Gatekeeper feature. Otherwise your system will report that the app is "damaged". First run this command in your Terminal (using your admin password when prompted):

`sudo spctl --master-disable`

Then open the app with Control+click in the Finder. 

Finally, run this command to secure your system again:

`sudo spctl --master-enable`

macumbista, 2017-10-16:
For the curious, I also tried this and it did not seem to add the security exception:

`sudo spctl --add /Applications/DIYLC.app`

Therefore I recommend the method of *temporarily* disabling Gatekeeper shown above.

Dadlyman, 2019-03-07:
If you move the DIYLC.app to the Applications folder then the exception will work.

Open finder, go to downloads, find the DIYLC.app and drag it to the Applications folder. 

sudo spctl --add /Applications/DIYLC.app

It works for me.

bancika, 2020-04-16:
Added to the manual, thanks

closed 2020-04-16 by bancika