bancika/diy-layout-creator #340: Trimmer type/size

chancelima, 2017-11-30:
Hi there,
What an amazing program this is, great work!!! I just have a question/request. I have the latest version 3.40.0. The trimmer takes up a lot of room. The trimmers I use at home in my projects are the same pin arrangements as in the program but the one in the program has a lot of overhang on all sides partially covering the surrounding tracks or adjacent components. The ones I (and others too) physically use don't have that overhang, they only take up the space occupied by the pins. Is it possible to have trimmers without the excess on the sides, or if there is such thing can you tell me how I can get it. It would just look more accurate and tidier, I have attached a picture for reference. Thanks so much. Chance.
[image]

bancika, 2017-11-30:
There's no such variant of trimmer yet, but I'll add it to the queue.

Cheers

EL34xyz, 2018-12-03:
Thanks for the suggestion. 
I also use small trimmer pots
The round trimmer body does not stick out beyond the 3 pins

The two front legs are 5mm center to center
The rear wiper leg is 10mm center to center with the two front legs
Here's an image of one I use

[image]

bancika, 2018-12-15:
added for 3.42

closed 2018-12-15 by bancika

EL34xyz, 2018-12-15:
Thanks Bancika

chancelima, 2018-12-16:
Wow that's great!
I think your software is number one, your work is much appreciated Bancika.

Chance