bancika/diy-layout-creator #343: DIYLC gets stuck when changing the project property

FredMercure, 2018-02-08:
Hello, since a couple of days i can't change the properties of my project whithin DIYLC. It get stuck before openning the panel to do so. here is the terminal output. 
_at java.lang.Thread.run(Thread.java:748)
01:15:11,808  INFO [AWT-EventQueue-0] ActionFactory - Edit Project triggered
01:15:11,816 DEBUG [AWT-EventQueue-0] PropertyEditorDialog - Creating property editor for: [Title = New Project, Description = null, Author = mint-mercure, Width = 29cm, Height = 21cm, Grid spacing = 0,1in, Default Font = java.awt.Font[family=Dialog,name=Tahoma,style=plain,size=14], Default Font Size = 14]_Then nothing no matter how long i'll wait.
i have changed the java management program to Oracle without luck.
Anyone can help me out?
Thanks.
-fM

bancika, 2018-02-08:
did it work before?

FredMercure, 2018-02-08:
That is a good question, but i guess it did.
Then it also freezes when i am adding text on the board. i'll try on an another ubuntu install to see if out at first run.

FredMercure, 2018-02-08:
Actually on the other computer it works, it takes time 10 seconds but it works.
So it comes probably from the Java thing, which seem to hang. i changed from the open source player to the oracle one without success. I realize it is not a bug as such but more of a problem in my system. may be it dosen't have to be reported here? Well if you have any idea what could go wrong i would be eternally grateful, Thanks.
FM

N9XYP, 2018-02-08:
I just started a new project using 3.40.0 and when changing a label it hangs for some time. I though that it was the latest version but it happens with the last two releases as well. The second time I changed a label the edit window opens like normal. If you restart diylc then it will do the same thing. I have Java 8 Update 161 (build 1.8.0_161-b12). Win7 Ulitmate SP1 on Intel i5-3570K with 16 GB RAM.

FredMercure, 2018-02-08:
Hey N9XYP, i try to wait untill it shows the panel and if it does. i'll report here. here is the java:
 java-8-oracle, and linux mint Thanks.

FredMercure, 2018-02-08:
Well, N9XYP you were right i had to wait a while not 20 minutes sure but enough to get me distracted. :)
But i have access to the panel, same thing happens when i create label.
Thanks.

bancika, 2018-12-22:
dupe of #372

closed 2018-12-22 by bancika

bancika, 2018-12-25:
Can anyone please try the latest release and see if it makes an improvement? It should

FredMercure, 2018-12-27:
Hello there, the new version looks good but unfortunately i have the same
behavior when changing the projects settings; it takes a while.
Thanks.