bancika/diy-layout-creator #344: possible tool suggestion

starpath07, 2018-02-18:
please consider the possible inclusion of a "path" tool, mostly in the strip- and printed circuit board mode in which whenever you click on a particular pcb track or stripboard conductor, these would be highlight the whole connected length clicked upon so that it would be easy to check for missing and/or misplaced cuts and/or copper strip bridging jumpers in stripboard mode or misrouting in pcb mode. This would help very much during board designing. Thanks and well done.

iiimonfiiire, 2018-02-19:
+1. This is a great idea. I've been doing this manually while creating vero layouts by adding different colored traces to the board. This feature would be a huge time-saver.

bancika, 2018-12-16:
done for 3.42.0

closed 2018-12-16 by bancika