bancika/diy-layout-creator #345: Export PDF Java issue

animalejo, 2018-02-27:
This message appears to me when I try to export to PDF, before I didn't have this problem and now out of nowhere it shows me that error :(

https://www.dropbox.com/s/vv0lvjbt3viav18/pdfjavaproblem.jpg?dl=0

bancika, 2018-02-27:
Can you please upload the latest log file? It should be here C:\Users\<your user>\.diylc\logs

animalejo, 2018-02-27:
this is the last log file:

> 00:53:33,053 DEBUG [main] DIYLCStarter - Java version: 1.8.0_161-b12 by Oracle Corporation
> 00:53:33,053 DEBUG [main] DIYLCStarter - OS: Windows 10 10.0
> 00:53:33,053  INFO [main] DIYLCStarter - Starting DIYLC with working directory C:\Program Files (x86)\DIYLC
> 00:53:33,340  INFO [main] ConfigurationManager - Initializing configuration
> 00:53:33,459  INFO [main] Presenter - installPlugin(ToolBox)
> 00:53:33,533  INFO [main] Presenter - Loading component types.
> 00:53:34,659 DEBUG [main] Presenter - Guitar: [Single Coil Pickup, P-90 Single Coil Pickup, Humbucker Pickup, LP Switch, Lever Switch]
> 00:53:34,665 DEBUG [main] Presenter - Semiconductors: [SIP IC, Transistor (TO-3 package), Transistor (TO-1 package), Diode (plastic), Transistor (TO-220 package), LED, Diode (glass), DIP IC, Transistor (TO-92 package)]
> 00:53:34,665 DEBUG [main] Presenter - Boards: [Marshall Style Perf Board, Terminal Strip, Vero Board, Blank Board, TriPad Board, Perf Board w/ Pads, Eyelet Board, Breadboard]
> 00:53:34,665 DEBUG [main] Presenter - Passive: [Film Capacitor (radial), Electrolytic Capacitor (radial), Ceramic Capacitor (radial), Potentiometer, Electrolytic Capacitor (axial), Trimmer Potentiometer, Resistor, Film Capacitor (axial)]
> 00:53:34,665 DEBUG [main] Presenter - Schematics: [Battery (symbol), JFET (symbol), Triode (symbol), Transformer Core (symbol), Resistor (symbol), MOSFET (symbol), IC (symbol), Diode (symbol), LED (symbol), Diode (symbol), Inductor (symbol), BJT (symbol), Ground (symbol), Schottky Siode (symbol), Capacitor (symbol), Zener Diode (symbol), Transformer Coil (symbol), Pentode (symbol), Potentiometer (symbol)]
> 00:53:34,666 DEBUG [main] Presenter - Electromechanical: [Closed 1/4" Jack, Mini Toggle Switch, Cliff 1/4" Jack, Mini Relay, 9V Battery Snap, Plastic DC Jack, Open 1/4" Jack]
> 00:53:34,666 DEBUG [main] Presenter - Misc: [Bill of Materials, Label, PCB Text, Image]
> 00:53:34,666 DEBUG [main] Presenter - Connectivity: [Dot, Jumper, Copper Trace, Trace Cut, Ground Fill, Hookup Wire, Solder Pad, Eyelet, Curved Trace]
> 00:53:34,666 DEBUG [main] Presenter - Tubes: [Tube Socket]
> 00:53:34,666 DEBUG [main] Presenter - Uncategorized: [ElectrolyticCanCapacitor]
> 00:53:34,666 DEBUG [main] Presenter - Shapes: [Line, Polygon, Ellipse, Rectangle]
> 00:53:34,666 DEBUG [main] Presenter - SMD: [SMD Resistor, SMD Capacitor]
> 00:53:34,746  INFO [main] MainFrame - injectGUIComponent(org.diylc.swing.plugins.toolbox.ComponentTabbedPane, 1)
> 00:53:34,776  INFO [main] Presenter - installPlugin(FileMenuPlugin)
> 00:53:34,778  INFO [main] MainFrame - injectMenuAction(New, File)
> 00:53:34,784  INFO [main] MainFrame - injectMenuAction(Open, File)
> 00:53:34,787  INFO [main] MainFrame - injectMenuAction(Import, File)
> 00:53:34,788  INFO [main] MainFrame - injectMenuAction(Save, File)
> 00:53:34,789  INFO [main] MainFrame - injectMenuAction(Save As, File)
> 00:53:34,790  INFO [main] MainFrame - injectDynamicSubmenu(Recent Files, icon, File)
> 00:53:34,791  INFO [main] MainFrame - injectMenuAction(Separator, File)
> 00:53:34,792  INFO [main] MainFrame - injectMenuAction(Export to PDF, File)
> 00:53:34,793  INFO [main] MainFrame - injectMenuAction(Export to PNG, File)
> 00:53:34,794  INFO [main] MainFrame - injectMenuAction(Print..., File)
> 00:53:34,795  INFO [main] MainFrame - injectSubmenu(Trace Mask, icon, File)
> 00:53:34,795  INFO [main] MainFrame - injectMenuAction(Export to PDF, Trace Mask)
> 00:53:34,796  INFO [main] MainFrame - injectMenuAction(Export to PNG, Trace Mask)
> 00:53:34,796  INFO [main] MainFrame - injectMenuAction(Print..., Trace Mask)
> 00:53:34,797  INFO [main] MainFrame - injectMenuAction(Create B.O.M., File)
> 00:53:34,797  INFO [main] MainFrame - injectMenuAction(Separator, File)
> 00:53:34,798  INFO [main] MainFrame - injectMenuAction(Exit, File)
> 00:53:34,803  INFO [main] Presenter - installPlugin(EditMenuPlugin)
> 00:53:34,803  INFO [main] MainFrame - injectMenuAction(Undo, Edit)
> 00:53:34,804  INFO [main] MainFrame - injectMenuAction(Redo, Edit)
> 00:53:34,804  INFO [main] MainFrame - injectMenuAction(Separator, Edit)
> 00:53:34,805  INFO [main] MainFrame - injectMenuAction(Cut, Edit)
> 00:53:34,806  INFO [main] MainFrame - injectMenuAction(Copy, Edit)
> 00:53:34,807  INFO [main] MainFrame - injectMenuAction(Paste, Edit)
> 00:53:34,808  INFO [main] MainFrame - injectMenuAction(Duplicate, Edit)
> 00:53:34,809  INFO [main] MainFrame - injectMenuAction(Separator, Edit)
> 00:53:34,810  INFO [main] MainFrame - injectMenuAction(Select All, Edit)
> 00:53:34,811  INFO [main] MainFrame - injectMenuAction(Edit Selection, Edit)
> 00:53:34,812  INFO [main] MainFrame - injectMenuAction(Delete Selection, Edit)
> 00:53:34,813  INFO [main] MainFrame - injectSubmenu(Transform Selection, icon, Edit)
> 00:53:34,814  INFO [main] MainFrame - injectMenuAction(Rotate Clockwise, Transform Selection)
> 00:53:34,815  INFO [main] MainFrame - injectMenuAction(Rotate Counterclockwise, Transform Selection)
> 00:53:34,815  INFO [main] MainFrame - injectMenuAction(Separator, Transform Selection)
> 00:53:34,816  INFO [main] MainFrame - injectMenuAction(Mirror Horizontally, Transform Selection)
> 00:53:34,817  INFO [main] MainFrame - injectMenuAction(Mirror Vertically, Transform Selection)
> 00:53:34,817  INFO [main] MainFrame - injectMenuAction(Separator, Transform Selection)
> 00:53:34,818  INFO [main] MainFrame - injectMenuAction(Nudge, Transform Selection)
[... cut ...]
> 	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
> 	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
> 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
> 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
> 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
> 	at java.awt.EventDispatchThread.run(Unknown Source)
> Caused by: java.lang.NullPointerException
> 01:05:12,019  INFO [AWT-EventQueue-0] ConfigurationManager - Saving configuration

animalejo, 2018-02-27:
now i see that's happen only with one project, i didn't make any particular work or something strange, just new project like any others. i don't know what happen...

bancika, 2018-02-27:
I see. When you open any other projects it works fine? If so, could you please upload the project itself, you can upload directly here. I'll debug on my side.

animalejo, 2018-02-27:
here you go:

https://www.dropbox.com/s/689f0a80ghkwkjs/bigmuffALT.diy?dl=0

Thanks

animalejo, 2018-02-27:
i forget....no problem with other projects

bancika, 2018-02-27:
interesting, it works fine in my DIYLC v40. Generated PDF with no issue. Which leads me to believe that there's something about the local configuration. Maybe it saved some bad setting or something. That's the only explanation I have. I assume that restart of the app doesn't help either. Could you upload config.xml from .diylc? Cheers

animalejo, 2018-02-27:
yeah, restart doesn't work, definility is only with that project :(

there you go: https://www.dropbox.com/s/zkjwfqm7zo73ft9/config.xml?dl=0

animalejo, 2018-02-27:
dude i solved the problem, it was the Font !!
i use this one: https://www.dropbox.com/s/ow5axgjyev3nndm/diylcissuefont.jpg?dl=0
if i delete it the problem is gone...
maybe the font is corrupted or something, i don't know

Thanks anyway
great app ! i'm wait for the new version jejeje
Cheers !!

bancika, 2018-02-27:
ha! Cool. I don't have that font so it worked here even with your config because it replaced that font with default. I'll check that font out and see what it does to the process.

Cheers

bancika, 2018-02-27:
btw, great looking layout!

closed 2018-02-27 by bancika

animalejo, 2018-02-27:
thanks dude

cheers from Chile