bancika/diy-layout-creator #348: BOM export to excel does not work

jojokeo, 2018-03-13:
I've tried naming the old way *.xls and the newest version *.xlsx - neither work!

bancika, 2018-03-13:
which OS are you using?

jojokeo, 2018-03-13:
Windows 7 professional

EL34xyz, 2018-12-03:
I can verify that BOM save to Excel .xls is not working for me either.
The save dialog box opens and I type a file name, but the file is never saved where I have specified
Save to csv, png and html all work fine for me

Windows 7 pro here also
I have Microsoft Office installed  on this desktop

bancika, 2018-12-16:
fixed in 3.42.0

closed 2018-12-16 by bancika

jojokeo, 2018-12-16:
Hello Bancika!

I only see version 3.41.1 available. Can you say when version 3.42.0 will become available?

Thank you!

Jojokeo

EL34xyz, 2018-12-16:
Bancika posted on the facebook DIYLC group that he is working on adding a few more fixes and new features before doing the next release
You should join the group here
https://www.facebook.com/groups/866347656903552/