bancika/diy-layout-creator #356: Idea: minimum distance (lines and pads)

divanorosso, 2018-06-14:
Hello,
I am used to using thicker lines, for example in the pads use 0.8 mm holes and pad size of 2.4 mm. The problem is that sometimes I find pads very close together. I almost always notice it but sometimes it escapes me and in the pcb I find the pads too close. So pcb goes on the trash and i must restart my work. Is it possible to insert a "minimum distance" in the settings? So when I try to put two pads too close the software blocks the operation. Thank you.

bancika, 2020-12-18:
After more than two years I still don't see a practical solution for this problem. There are two issues

1) it's bloody difficult to compute the distance between two areas of arbitrary shapes containing lines, polygons, circles that can be placed on arbitrary place on the canvas and also be combined together into more complicated shapes
2) it's not possible to block moving traces around because in some cases we want to connect two traces together, e.g. traces that go to one main ground bus or V+ bus. The best we could do is to somehow highlight traces that are problematic, but even that is very hard because of 1)

bancika, 2020-12-20:
Actually, thought on an idea that could work for #1...will add something in 4.13

closed 2020-12-20 by bancika