bancika/diy-layout-creator #364: Mac OSX 10.12.6 'Sierra' cannot open App

SonusMagus, 2018-09-26:
I am getting the malicious app warning and being told to move the App to the Trash. 

I have followed instructions to mark the App as excluded and not getting any joy

It will not start from run.sh either

Any advice appreciated

Thanks

jonathanfann, 2018-10-06:
I encountered this same issue.  I ended up creating an Automator Application that ran a shell script: 

`open /{{ path to the app }}/DIYLC\ 2.app/Contents/Resources/Java/diylc.jar`

the first time you need to actually right click the file there ^ and hit open (this allows you to run it in the automator script...).  sorry that's kinda complicated.  

anyways, it does a little pop up like "you shouldn't open this this way" (from the app, not from MacOS), but click yes and you'll be making stripboard layouts on mac again.  but yeah, this is a pretty bad workaround, but if you're willing to put in the time to set it up, it's not so bad once you get it rolling.

I even added a stupid icon to mine and plopped it in whatever that bar at the bottom is called: 
[image]

SonusMagus, 2018-10-07:
thanks Jonathan that works cleaner than what I have been doing ;/)
so I wonder if there's an advantage over simply dragging the .jar out of the broken App and double clicking it?
thanks

bancika, 2018-12-25:
There is an advantage. Double clicking the .jar starts the application without the recommended memory settings and potentially other stuff. It'll work either way, though.

martinloren, 2019-01-31:
I made a package to fix the opening issue n Mac OSX. For future release just update the .jar file inside and eventually other contents. The framework will continue to work.
Issue:
- it still show the message to do not open it directly by the .jar file. Just need to click OK each time you open the app.
- Once opened it show version 0.0 (a bug in the .jar?) so it always says that there is a new release to download.

[diylc.app.zip](https://github.com/bancika/diy-layout-creator/files/2816330/diylc.app.zip)

bancika, 2019-01-31:
Thanks!

Hmm, the version is read from update.xml file, which probably means that it doesn't have correct access to that file. Also, do you see factory variants? For example, IC component should have 2 of them. Those are stored in variants.xml file. It's possible that that's another issue.

What did you use to generate the app? I have automated process for building deploy packages for all platforms, wondering how I could automate this. What's different compared to my app package?

Thanks!

martinloren, 2019-01-31:
This is the tool that works: jar2app 1.0.1 (https://github.com/Jorl17/jar2app)
I've just generated an app with just the .jar inside, then in Mac OS is possible to open the .app packages and drop inside the files/modify, so I put all other files and it worked. I'm trying to load the XML changing folders inside the package. I see on the app title bar it show version 0.0 but in About box it show version 3.0.0.

The .xml file is also the latest one but inside it report:

			<versionNumber>
				<major>3</major>
				<minor>0</minor>
				<build>4</build>
			</versionNumber>

martinloren, 2019-01-31:
This version should show also App Icon (I just put in the right folder linking with the .pinfo file).
Variants XML works fine, I tried to move out and it don't work, move in and it show all variants. Version XML is in the same folder of Variants XML but still have that issue.

[DIYLC.app.zip](https://github.com/bancika/diy-layout-creator/files/2816522/DIYLC.app.zip)

bancika, 2019-01-31:
Sorry for asking, I'm not a mac user so not sure how everything works. Which issue is this build solving? I tried the "official" OSX build on a sierra and once you allow it in the gatekeeper, it runs fine. Is there any other issue I'm not aware of?
Tnx again

martinloren, 2019-01-31:
I've tried several releases for Mac OSX and none could start even if authorized. I've always used the trick to open the .jar before.

Here the screenshot of the message that appean when you try to open the app on Mac. (Click on the App and then -> Open)
[image]

bancika, 2019-02-11:
I use eclipse. Tried replacing the jars in the zip with the latest build but getting this error on my sierra VM

[image]

spyrosze, 2019-05-23:
I faced the same app warning on Mac OS 10.14.5 Mojave and had to run the terminal trick to open the app. Type this in terminal. 
`xattr -cr /path/to/app`

@bancika this tool seems really *fantastic*  congrats! I am familiar with the drawings as a DIY effects builder and looking forward to use it to draw schematics. I am a UX designer by trade and I'll be more than happy to provide feedback when I get to use it. One thing that seems strange is the pixelated font on my [Mac](https://www.dropbox.com/s/1ady85cgxjtj1x1/Screenshot%202019-05-23%2016.21.37.png?dl=0).

closed 2020-03-26 by bancika