bancika/diy-layout-creator #370: Will mac version be updated to Java 8?

PhotonWill, 2018-10-27:
Just wondering before install Java 6 on my Mac, any chance a new OS X build will be available for Java 8 (Mojave).

EL34xyz, 2018-12-19:
Announcement in DIYLC

[image]

JackMillerSD, 2018-12-26:
Followed instructions here (https://oliverdowling.com.au/2014/03/28/java-se-8-on-mac-os-x/) and DIYLC seems to work just fine.

PhotonWill, 2018-12-27:
Did you install the JRE or JDK? 
Oliver actually references a newer method at:
https://oliverdowling.com.au/2015/10/09/oracles-jre-8-on-mac-os-x-el-capitan/

Which is actually required for diy-layout, I assume just the JRE?
What version Mac OS are you on? I'm using 10.14.2

JackMillerSD, 2018-12-28:
> Did you install the JRE or JDK?
> Oliver actually references a newer method at:
> https://oliverdowling.com.au/2015/10/09/oracles-jre-8-on-mac-os-x-el-capitan/
> 
> Which is actually required for diy-layout, I assume just the JRE?
> What version Mac OS are you on? I'm using 10.14.2

Installed the JDK.
Mac OS version is 10.13.6 (High Sierra)

sdmeyers, 2019-01-11:
FYI... To get DIYLC to work on Mac OS X (tested on the latest Mojave) follow these steps.

Completely remove any current java versions:
In Terminal run:
```
sudo rm -fr /Library/Internet\ Plug-Ins/JavaAppletPlugin.plugin 
sudo rm -fr /Library/PreferencePanes/JavaControlPanel.prefPane 
sudo rm -fr ~/Library/Application\ Support/Oracle/Java
```
to remove any JRE installed.
(details: https://www.java.com/en/download/help/mac_uninstall_java.xml )
Then continuing in terminal type:
```
cd /Library/Java/JavaVirtualMachines/
sudo rm -rf *.jdk
```
(details: https://docs.oracle.com/javase/8/docs/technotes/guides/install/mac_jdk.html#A1096903 )

2. Install java 8 using "homebrew" (installing with homebrew fixes some issues automatically, for example, it will stop the Java 6 warning):

First, if you don't have homebrew installed, install it (again... in Terminal):
`/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"`

(details: https://brew.sh )

Next to install Java8 (specifically... you do not want the latest version of Java (11) as things are very different and this app will not work) type this is Terminal.
```
brew tap caskroom/versions
brew cask install java8
```

Now in Terminal type:
`java -version`

and assuming everything is good you should get the following output (or similar):
```
java version "1.8.0_192"
Java(TM) SE Runtime Environment (build 1.8.0_192-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.192-b12, mixed mode)

```

3. Add a gatekeeper exception for DIYLC.app (Gatekeeper is a security mechanism in OS X that prevents any unsigned apps from launching. In recent versions of Mac OS there is no longer a visible option to bypass it entirely, but, at least for now if you know the option you can bypass it for certain apps). Using Terminal (still)...
` sudo xattr -rd com.apple.quarantine /Applications/DIYLC.app`

This assumes DIYLC.app is installed in you Applications directory. If it's somewhere else (i.e. `/Users/`_YourHome_`/Applications/DIYLC.app` ) use that path.

4. That should be it (works for me)... Happy DIYLCing.

mihalski, 2019-04-09:
I've followed the instructions provided by @sdmeyers but I still get "To open “DIYLC” you need to install the legacy Java SE 6 runtime."

Is there anything else that might need to be done?

Thank you in advance.

sdmeyers, 2019-04-10:
Yea. I noticed this too. The solution is easy(ish) though. I'll let you know what I ended up doing.

First, I uninstalled brew's java8 version and installed the latest java version instead. (This allows me to work on other modern Java stuff). `brew cask uninstall java8` followed by `brew cask install java`. As of this writing, that will install Java 12.

Next, when you launch DIYLC in the "To open “DIYLC” you need to install the legacy Java SE 6 runtime." dialog box click [More Info...]. This should take you to this page: https://support.apple.com/kb/DL1572?locale=en_US 

From there download the "legacy" version of Java and install it.

This will install a Java 1.6.0 runtime along with any other "newer" java runtime version you have installed. For example in my `/Library/Java/JavaVirtualMachines/` directory I have both `1.6.0.jdk` and `openjdk-12.jdk`. The command line version is still 12. `java -version` returns:

    openjdk version "12" 2019-03-19
    OpenJDK Runtime Environment (build 12+33)
    OpenJDK 64-Bit Server VM (build 12+33, mixed mode, sharing)

DIYLC will use Java 1.6.

(You still need to run `sudo xattr -rd com.apple.quarantine _(path to)_/DIYLC.app` every time you install or update DIYLC).

aguynamedben, 2020-03-21:
> From there download the "legacy" version of Java and install it.
>
> This will install a Java 1.6.0 runtime along with any other "newer" java runtime version you have installed.

FYI I was trying to follow these instructions to get DIYLC working on the current macOS version (Catalina, 10.15) and it seems macOS (or the installer?) allows having both installed.

Do you know of any way around this?

[image]

aguynamedben, 2020-03-21:
This is addresses in more detail in this duplicate issue: https://github.com/bancika/diy-layout-creator/issues/479

bancika, 2020-03-21:
I'm currently working on building DIYLC using appbuilder and Java 8. I'm almost there, but struggling with final steps. The app starts, but no window is shown. The icon appears in the task bar (or whatever is called on OSX) and I get some generic menu bar titled DIYLC, but with items that make no sense to me. It's not the real DIYLC menu.

If anyone knows what could be the issue, it'd be great. I have no experience with OSX. I can attach Info.plist file if needed and even the whole build.

[image]

bancika, 2020-03-22:
Can anyone give this version a try? I tried it on Mojave VM and it works fine, but it dies after starting on Sierra VM. Can't figure out the reason yet.

[diylc-4.0.1-osx.zip](https://github.com/bancika/diy-layout-creator/files/4365490/diylc-4.0.1-osx.zip)

bancika, 2020-03-26:
Released 4.0.1 with Java 8

closed 2020-03-26 by bancika