bancika/diy-layout-creator #372: Application Freeze on Project Settings

MoxorTheOne, 2018-10-29:
Version: 3.40.0
Platform: Windows 10
Steps:
1. Open DIYLC
2. Click Edit
3. Click Edit Project Settings
Result:
The application cease to respond (Java process makes considerable  CPU usage)

shouse1953, 2018-11-11:
I found that this occurred to me as well.  Being new to the program I paused for a bit,  After roughly 30 seconds the Project Settings dialog box appeared so my thought now is it was initializing and is a bit slow in doing so.  I haven't reviewed the code yet but will say if it is instantiating many Java objects that would explain the delay.  Even on a very fast system there is a lot of overhead in Java object instantiation.

blackcorvo, 2018-11-13:
It also happens when you first try to edit a text label.

bancika, 2018-11-13:
I think it has to do with font initialization. The first time you open
anything font-related it will hang while rendering sample for all fonts in
the system. I'll see what can be done to simplify it or optimize.
Cheers

noirimipsum, 2018-11-23:
confirmed delay here as well

animalejo, 2018-11-27:
here to......same issue. freeze all time

bancika, 2018-12-12:
interesting, it takes 2-3 seconds on all machines I have access to (win10 and win7). How many fonts do you have installed?

bancika, 2018-12-22:
Yeah, it's definitely fonts. I'm not sure what makes it work so slow on some machine but not others. Maybe it's the number of fonts, or some fonts in particular make it slow to render.
I could add a configuration that disables font samples in the font picker...that way people with problems can just disable font preview, but then they need to know how each font looks, which is not ideal.

bancika, 2018-12-25:
Can anyone please try the latest release and see if it makes an improvement? It should

noirimipsum, 2018-12-28:
confirmed fixed on my ubuntu xenial/mint sylvia

closed 2019-01-03 by bancika