bancika/diy-layout-creator #373: Grid has disappeared?!?!?!

Curranmurphy, 2018-11-15:
So the grid has disappeared, I have uninstalled and re-installed but the grid is still missing. I kind of like having the grid in my project window. the snap still works. I am using DIYLC on windows 7 32bit

bancika, 2018-11-15:
did you try changing the "Show Grid" option from the menu?

Curranmurphy, 2018-11-15:
yes sir I did, the show grid is checked and the grid size is (.1) the weird
thing is that I have the grid on other computers I use this program on. AND
I had the grid on here before and it just "magic'ed" away for no reason. I
have also opened up several other project with the same results. very
odd......

-- 
Curran Murphy

Owner, Builder for Custom Audio Mutation Amplification and Guitars

Owner, Producer, Engineer for Custom Audio Mutation Studios

234-788-6141

This email and any files transmitted with it are confidential and
intended solely for the use of the individual or entity to whom they
are addressed. If you have received this email in error please notify
the system manager. This message contains confidential information and
is intended only for the individual named. If you are not the named
addressee you should not disseminate, distribute or copy this e-mail.
Please notify the sender immediately by e-mail if you have received
this e-mail by mistake and delete this e-mail from your system. If you
are not the intended recipient you are notified that disclosing,
copying, distributing or taking any action in reliance on the contents
of this information is strictly prohibited.

bancika, 2018-12-15:
can you please post your config file. It should be in c:\users\<your name>\.diylc

EL34xyz, 2018-12-18:
I think you can close this Bancika
Curran reported this on FB and does not seem to have the issue any more

closed 2018-12-18 by bancika