bancika/diy-layout-creator #375: Graphic problem

animalejo, 2018-11-24:
Recently started having this problem, when I opened menus it would seem that everything is graphically unconfigured.
It only happens with this program, with others I do not have any kind of problem, I am using a new laptop, with an 8th generation i7 processor, 16gb ram and 8gb nvidia1070 video card, so I discard it as something from the graphics card.

I forgot to comment that this happens with any project, not only the one that appears in the video.

Video: https://drive.google.com/open?id=1k3yWuHTJ4AcYiI42GfrBahl6awdKvu6Q

Help me mr Bancika !

animalejo, 2018-11-24:
Aqui otro video mostrando otro proyecto, sucede lo mismo.
ya he actualizado Java pero no arregla nada y tambien reinstale el programa pero pasa lo mismo. Uso la version 3.40

VIDEO 2 : https://drive.google.com/open?id=1NC6GsD1GqjB8Evm5skpXjGK2h16lhqQa

animalejo, 2018-11-27:
please i need fix this!! i can't do anything :( !!!!

bancika, 2018-11-27:
It's likely that some java update messed things up. What is your java version? Can you please include the latest log file from c:\Users\<your name>\.diylc\log

animalejo, 2018-11-27:
Hi !!

i have the version 4 with the last update 191

here my log file:

11:37:25,599 DEBUG [main] DIYLCStarter - Java version: 1.8.0_191-b12 by Oracle Corporation
11:37:25,600 DEBUG [main] DIYLCStarter - OS: Windows 10 10.0
11:37:25,600  INFO [main] DIYLCStarter - Starting DIYLC with working directory C:\Program Files (x86)\DIYLC
11:37:26,355  INFO [main] ConfigurationManager - Initializing configuration
11:37:26,450  INFO [main] Presenter - installPlugin(ToolBox)
11:37:26,512  INFO [main] Presenter - Loading component types.
11:37:26,672 DEBUG [main] Presenter - Guitar: [Lever Switch, Single Coil Pickup, LP Switch, Humbucker Pickup, P-90 Single Coil Pickup]
11:37:26,674 DEBUG [main] Presenter - Passive: [Potentiometer, Ceramic Capacitor (radial), Electrolytic Capacitor (radial), Film Capacitor (radial), Trimmer Potentiometer, Electrolytic Capacitor (axial), Film Capacitor (axial), Resistor]
11:37:26,674 DEBUG [main] Presenter - Boards: [Breadboard, Marshall Style Perf Board, Eyelet Board, Vero Board, Blank Board, TriPad Board, Terminal Strip, Perf Board w/ Pads]
11:37:26,674 DEBUG [main] Presenter - Semiconductors: [LED, Diode (glass), Transistor (TO-92 package), DIP IC, Transistor (TO-220 package), Transistor (TO-1 package), Transistor (TO-3 package), Diode (plastic), SIP IC]
11:37:26,674 DEBUG [main] Presenter - Schematics: [Diode (symbol), Inductor (symbol), Zener Diode (symbol), Transformer Coil (symbol), Pentode (symbol), Transformer Core (symbol), Diode (symbol), Ground (symbol), Schottky Siode (symbol), Triode (symbol), MOSFET (symbol), Resistor (symbol), IC (symbol), Battery (symbol), LED (symbol), Potentiometer (symbol), Capacitor (symbol), BJT (symbol), JFET (symbol)]
11:37:26,674 DEBUG [main] Presenter - Electromechanical: [Closed 1/4" Jack, Plastic DC Jack, Mini Relay, Mini Toggle Switch, Cliff 1/4" Jack, 9V Battery Snap, Open 1/4" Jack]
11:37:26,674 DEBUG [main] Presenter - Misc: [Bill of Materials, Image, Label, PCB Text]
11:37:26,674 DEBUG [main] Presenter - Connectivity: [Jumper, Copper Trace, Ground Fill, Hookup Wire, Dot, Trace Cut, Eyelet, Curved Trace, Solder Pad]
11:37:26,674 DEBUG [main] Presenter - Tubes: [Tube Socket]
11:37:26,674 DEBUG [main] Presenter - Uncategorized: [ElectrolyticCanCapacitor]
11:37:26,674 DEBUG [main] Presenter - Shapes: [Ellipse, Line, Polygon, Rectangle]
11:37:26,674 DEBUG [main] Presenter - SMD: [SMD Capacitor, SMD Resistor]
11:37:26,753  INFO [main] MainFrame - injectGUIComponent(org.diylc.swing.plugins.toolbox.ComponentTabbedPane, 1)
11:37:26,782  INFO [main] Presenter - installPlugin(FileMenuPlugin)
11:37:26,785  INFO [main] MainFrame - injectMenuAction(New, File)
11:37:26,790  INFO [main] MainFrame - injectMenuAction(Open, File)
11:37:26,792  INFO [main] MainFrame - injectMenuAction(Import, File)
11:37:26,793  INFO [main] MainFrame - injectMenuAction(Save, File)
11:37:26,793  INFO [main] MainFrame - injectMenuAction(Save As, File)
11:37:26,794  INFO [main] MainFrame - injectDynamicSubmenu(Recent Files, icon, File)
11:37:26,795  INFO [main] MainFrame - injectMenuAction(Separator, File)
11:37:26,796  INFO [main] MainFrame - injectMenuAction(Export to PDF, File)
11:37:26,797  INFO [main] MainFrame - injectMenuAction(Export to PNG, File)
11:37:26,798  INFO [main] MainFrame - injectMenuAction(Print..., File)
11:37:26,798  INFO [main] MainFrame - injectSubmenu(Trace Mask, icon, File)
11:37:26,799  INFO [main] MainFrame - injectMenuAction(Export to PDF, Trace Mask)
11:37:26,799  INFO [main] MainFrame - injectMenuAction(Export to PNG, Trace Mask)
11:37:26,799  INFO [main] MainFrame - injectMenuAction(Print..., Trace Mask)
11:37:26,800  INFO [main] MainFrame - injectMenuAction(Create B.O.M., File)
11:37:26,800  INFO [main] MainFrame - injectMenuAction(Separator, File)
11:37:26,801  INFO [main] MainFrame - injectMenuAction(Exit, File)
11:37:26,804  INFO [main] Presenter - installPlugin(EditMenuPlugin)
11:37:26,804  INFO [main] MainFrame - injectMenuAction(Undo, Edit)
11:37:26,805  INFO [main] MainFrame - injectMenuAction(Redo, Edit)
11:37:26,805  INFO [main] MainFrame - injectMenuAction(Separator, Edit)
11:37:26,806  INFO [main] MainFrame - injectMenuAction(Cut, Edit)
11:37:26,807  INFO [main] MainFrame - injectMenuAction(Copy, Edit)
11:37:26,808  INFO [main] MainFrame - injectMenuAction(Paste, Edit)
11:37:26,808  INFO [main] MainFrame - injectMenuAction(Duplicate, Edit)
11:37:26,809  INFO [main] MainFrame - injectMenuAction(Separator, Edit)
11:37:26,809  INFO [main] MainFrame - injectMenuAction(Select All, Edit)
11:37:26,810  INFO [main] MainFrame - injectMenuAction(Edit Selection, Edit)
11:37:26,811  INFO [main] MainFrame - injectMenuAction(Delete Selection, Edit)
11:37:26,812  INFO [main] MainFrame - injectSubmenu(Transform Selection, icon, Edit)
11:37:26,812  INFO [main] MainFrame - injectMenuAction(Rotate Clockwise, Transform Selection)
11:37:26,813  INFO [main] MainFrame - injectMenuAction(Rotate Counterclockwise, Transform Selection)
11:37:26,813  INFO [main] MainFrame - injectMenuAction(Separator, Transform Selection)
11:37:26,814  INFO [main] MainFrame - injectMenuAction(Mirror Horizontally, Transform Selection)
11:37:26,815  INFO [main] MainFrame - injectMenuAction(Mirror Vertically, Transform Selection)
11:37:26,815  INFO [main] MainFrame - injectMenuAction(Separator, Transform Selection)
11:37:26,816  INFO [main] MainFrame - injectMenuAction(Nudge, Transform Selection)
[... cut ...]
	at org.diylc.swing.plugins.statusbar.StatusBar$1.doInBackground(Unknown Source)
	at org.diylc.swing.gui.MainFrame$4.doInBackground(Unknown Source)
	at javax.swing.SwingWorker$1.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at javax.swing.SwingWorker.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
11:37:29,825  INFO [SwingWorker-pool-1-thread-3] UpdateChecker - 0 updates found

animalejo, 2018-12-15:
I still have this problem, it's so frustrating.
I can not work on anything, I have stopped using the software.

So sad...

bancika, 2018-12-15:
did you try installing java 6?

bancika, 2018-12-28:
> Last version, same issiue :( !

Nothing was done to address the issue. Did you try to use older java
version? 7 or even 6?

animalejo, 2019-01-05:
yes and it's the same :(
Even i uninstalled the last Java version and installed 6 version

https://drive.google.com/file/d/1NvdmYdcexvO-c8PV4CViMx1JYtQHrZLD/view?usp=sharing

bancika, 2019-01-05:
Ok, can you try to replace diylc.l4j.ini file in DIYLC installation folder with one of the attached files and see if that helps? I'm not sure what is wrong so I created 4 versions. Try each one separately.

[diylc.l4j.ini2.txt](https://github.com/bancika/diy-layout-creator/files/2729879/diylc.l4j.ini2.txt)
[diylc.l4j.ini3.txt](https://github.com/bancika/diy-layout-creator/files/2729880/diylc.l4j.ini3.txt)
[diylc.l4j.ini4.txt](https://github.com/bancika/diy-layout-creator/files/2729881/diylc.l4j.ini4.txt)
[diylc.l4j.ini1.txt](https://github.com/bancika/diy-layout-creator/files/2729882/diylc.l4j.ini1.txt)

animalejo, 2019-01-09:
seems like fixed the problem!!!

i tried first with "diylc.l4j.ini 2" and still bugging
but with the others it's looks like all work perfectly
i updated java to last version and still function ok

Maybe a new update will come from this jejejej

thanks man !!!!!!!!!

bancika, 2019-01-10:
Not sure if I should include this in the official build. I was experimenting with turning OpenGL and DirectX rendering on and off. It might affect performance and maybe cause issues on other machines. It's good to know that it helps, in case someone else has a similar issue.
Cheers

closed 2019-01-10 by bancika

Mortifly, 2019-01-11:
I had this problem too. diylc.l4j.ini1.txt works for me
Thanks

bancika, 2019-01-11:
Good to know, thanks for letting me know

Relithe, 2021-11-20:
Had same issue with the 4.2 release. Didn't come up until I got a new monitor - which is bizarre, but I'm not a programmer so maybe it's a common thing? I went from a 1920x1080 75 Hz monitor to a 2560x1440 FreeSync monitor with HDR. It only blanked out when I had the program open on the FreeSync monitor, but when I dragged the program window to my second monitor (1920x1080, 60 Hz) it would work normally. 

I replaced the .ini with #3, and it solved my issue. Just wanted to let you know.