bancika/diy-layout-creator #377: Clarify licensing

heuermh, 2018-11-30:
Hello, I've been a user of v2 of diylc for some time, and am looking forward to contributing to v3 here on Github.

I am a bit concerned though that while the README says "[S]ource code is released under GPL license" and links to the [Quick Guide to GPLv3](http://www.gnu.org/licenses/quick-guide-gplv3.html), as far as I can tell the GPL license has not been applied to the source code appropriately.

See

How to use GNU licenses for your own software
https://www.gnu.org/licenses/gpl-howto.html

Specifically for the repository here:
 - [ ] Add a COPYING file with a copy of the GNU GPL version 3
 - [ ] Put a license notice in each file
 - [ ] Optionally, add a startup license notice to the splash screen and About dialog

I can help with this via pull request.

Note also that a [non-commercial clause is incompatible with the GPL](https://www.gnu.org/licenses/gpl-faq.html#NoMilitary).

"I'd like to license my code under the GPL, but I'd also like to make it clear that it can't be used for military and/or commercial uses. Can I do this?

No, because those two goals contradict each other. The GNU GPL is designed specifically to prevent the addition of further restrictions."

IANAL, but I think what you want is to dual-license, Free Software under the GPL and a separate commercial use license.

bancika, 2018-12-03:
Hi Michael,

I know very little about licensing, in my professional career it's always someone else's business :)
Maybe LGPL would be a better choice here?

Either way, I wouldn't spend too much time, going through all files is definitely not worth the effort. I could do something useful for the app the users for that time.

Cheers

heuermh, 2018-12-03:
Hello @bancika, thank you for the quick response!

It would be no trouble for me to create a pull request that adds the COPYING file, puts the license notice in each file, and updates the README.

closed 2018-12-04 by bancika

rockola, 2018-12-09:
The wording in the README is still contradictory to the GPL: as it stands now there is an additional restriction (no commercial use), which is not allowed. GPL specifically allows commercial use, but the source to any changes must be released. If the intent is to say that commercial use without complying to the GPL is possible with another license (ie. dual licensing), then this should be stated.

My suggestion:

Source code is released under GNU General Public License version 3. Contact the project owner to obtain a license in case you plan to use the source code, or any part of it, without complying to the restrictions of GPLv3.

PR #378 (2018-12-04): Apply GPL version 3 license
Fixes #377