bancika/diy-layout-creator #380: Feature request - Dashed/Dotted Hookup Wire

EL34xyz, 2018-12-16:
This is just future feature request
Many people in the Tube amp community use dashed or dotted wires to indicate that a wire runs under a board as a jumper wire.
I do this in MS Visio for all my Hoffman turret board layouts

It would be nice to be able to do this in DIYLC also

I attached a screen shot showing what these under board jumper wires look like in Visio
Thanks Bancika

[image]

bancika, 2018-12-17:
[diylc-3.43.0.zip](https://github.com/bancika/diy-layout-creator/files/2688071/diylc-3.43.0.zip)

kroma-1, 2018-12-17:
Dotted or dashed lines aren`t working for me. The options are there, but nothing changes.

EL34xyz, 2018-12-17:
Dotted and dashed lines work fine for me in  windows 7 pro

[image]

EL34xyz, 2018-12-17:
bancika is still testing several line features with me. 
Hang loose while we work some stuff out

bancika, 2018-12-17:
releasing as part of 3.43

closed 2018-12-17 by bancika