bancika/diy-layout-creator #382: Feature request-Add a new tool bar

EL34xyz, 2018-12-19:
I use several of the Transform selection menu tools so much I was thinking how nice it would be to add them to a new toolbar

I use the keyboard shortcuts, but it's way faster to use an icon in the toolbar
It's faster than taking my hands off the mouse to do a two handed keyboard shortcut

I am so used to Visio, and a few other programs I use all the time, having these quick access toolbars
I pasted up something just showing what this toolbar might look like

Just a feature suggestion at this point
Chime in if you guys would be interested in this

[image]

kroma-1, 2019-01-18:
I like it.

bancika, 2019-03-22:
added for 3.57

[image]

closed 2019-03-22 by bancika

EL34xyz, 2019-03-23:
Cool, when is 3.57 coming out?

bancika, 2019-03-23:
Prolly this weekend.