bancika/diy-layout-creator #384: Curved traces are too round

gtranter, 2018-12-23:
Since v3.43.0 the corner radius of curved traces with three points (four points as well) has increased a great deal. This causes them to be too close to things - usually pads - that I am trying to route around and makes them unusable. Pretty much all of my layouts are no longer usable as a result. Please restore the original radius. Attached is a small example where curved traces have become unusable around square pads in my layouts.

gtranter, 2018-12-23:
The attachment feature didn't work so here's a link to the example image: https://github.com/bancika/diy-layout-creator/issues/384.

gtranter, 2018-12-23:
The attachment and edit features won't work so here's a link to the example image: https://imgur.com/a/OY9f4tK.

bancika, 2018-12-23:
Thanks for reporting. But are you sure about four-point curve?
I can see that I changed the behavior of 3 point curves and I'll change that for the next release. But don't think it's there for four-point curv.e
Cheers

bancika, 2018-12-25:
Can you please try the latest release and see if it works ok for you? It should

closed 2018-12-25 by bancika