bancika/diy-layout-creator #385: Mutiple Section On OSX Broken

JackMillerSD, 2018-12-25:
Can no longer select multiple items while holding down the 'command' key. Problem introduced in version 3.45.0

bancika, 2018-12-25:
Fixed for the next version.

iiimonfiiire, 2018-12-26:
This issue still reproduces with v3.46
Mac OS 10.14.1

JackMillerSD, 2018-12-26:
> This issue still reproduces with v3.46
> Mac OS 10.14.1

Yes, the issue is still not resolved

bancika, 2018-12-26:
Yes, I unknowingly switched to control. I have a MacOS VM using standard windows keyboard, and that was the only way I could test that it works.

JackMillerSD, 2018-12-26:
> Hmm, maybe I should revert part of this change and go back to Command key. I figured out a way to do it while still fixing the original issue with variants and whatnot. I'll post a patch release

Cool - thank-you for finding and fixing the issue.

JackMillerSD, 2018-12-26:
> Please try 3.46.1

How does one get the patched release?

bancika, 2018-12-26:
The same place

JackMillerSD, 2018-12-26:
> The same place
> […](#)
> -- Branislav Stojkovic [www.diy-fever.com](http://www.diy-fever.com)

Multiple items can now be selected by holding down the 'command' key.

bancika, 2018-12-26:
Lovely, so no issue there?

iiimonfiiire, 2018-12-27:
Verified fixed from my end. Thanks!