bancika/diy-layout-creator #391: 3.46.1 - Mouse roller zoom not working

EL34xyz, 2018-12-29:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates

EL34xyz, 2018-12-29:
Windows 7 pro
Just DL'd 3.46.1 this morning 
First thing I noticed is that the mouse wheel does not zoom in or out

Rolling the mouse wheel moves the workspace up and down
It will not zoom in or out

These two features are working
Holding Shift and rolling the mouse wheel moves the workspace Left and Right
Holding control and rolling the mouse wheel moves the workspace Left and Right

EL34xyz, 2019-01-01:
I went back and installed 46.0 for Windows
The mouse wheel zoom works in 46.0

Mouse wheel zoom does not work  in 46.1

bancika, 2019-01-01:
It works for me. Strange

EL34xyz, 2019-01-01:
Here's a video I did comparing mouse wheel zoom on 46.0 and 46.1

https://youtu.be/dNzvfD_9fKQ

EL34xyz, 2019-01-01:
Just tested 46.1 on a different desktop with Windows 10
Same thing
Mouse wheel moves the screen up and down - No zoom

bancika, 2019-01-01:
got it...I thought you meant ctrl+wheel.

closed 2019-01-08 by bancika