bancika/diy-layout-creator #392: Theme Removed from Config menu?

EL34xyz, 2018-12-30:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates

EL34xyz, 2018-12-30:
Theme selection is gone from Config menu in 3.46.1
Just curious why this was removed?

N9XYP, 2018-12-30:
Theme selection is there for me. DIYLC 3.46.1 on UbuntuStudio 32-bit.

EL34xyz, 2018-12-30:
Missing from my menu in win 7 pro
It's there in 45.0
Gone in 46.1

[image]

stinga11, 2019-01-02:
I have the same problem but I am using Windows 10.

bancika, 2019-01-02:
I think that themes have been out of windows installer forever. They are in
the zip, so you may have gotten them by extracting the zip for the old
version.
I'll fix in the next release.

EL34xyz, 2019-01-02:
Thanks
That makes sense because by me testing two versions, one is the .zip and one is the .exe install

closed 2019-01-08 by bancika