bancika/diy-layout-creator #393: [Product Request] Mobile version (iOS & Android)

iiimonfiiire, 2019-01-01:
A bit of a long shot but I don’t see another place on this platform to make such a request. In any case, I realize this is much bigger than a simple feature request - but I love this software and would love to use it on the go. Are there any plans for making this software available on mobile platforms?

ChrisOfBristol, 2019-01-08:
The process for submitting an application to Debian in order to make DIYLC available on Linux is arcane and bureacratic, and GetDeb an alternative which hosted user-submitted applications appears to have retired. There is now a better way of distributing software called a Snap. I have created one for DIYLC, but have got stuck as I don't know Java, I think it's a configuration file access problem. Have error log - could you have a look?

EL34xyz, 2019-02-07:
Just my .02
I personally would never use DIYLC on a mobile device
My desktops run circles around my Android tablet and phone

closed 2019-03-22 by bancika