bancika/diy-layout-creator #394: OSX Shared variants Request

MacFly77, 2019-01-06:
Sorry if posting in the wrong section.
This is not a bug!

After the recent update for OS X I'm very happy to use the variants in DIYLC.
However I use DIYLC on 2 different machines running both OS X.

I've searched where the variants might be stored so that I can alias them to my iCloud and using the same variants on both machines. However I am not able to locate them. 
Is there a way to share them? 
I know they should be stored somewhere locally on the machine as moving the DIYLC app to iCloud loads the app but not with the same variants.

It would be very useful, but I am already very happy with variants being possible on OS X.
Thank you

bancika, 2019-01-06:
Hi,
I plan to make it possible to have local and shared variants, so I can ship DIYLC with some variants from the "factory". In the meantime, the only way for you to do it is to copy that part of the config file from one machine to the other...or take the whole config file for that matter. It's in user directory under .diylc
Cheers

MacFly77, 2019-01-06:
Great to hear you are working on this.

I assume the location is for Windows. I can not find the location on Mac OSX El Capitain.

MacFly77, 2019-01-06:
Sorry Found it....needed to see mu hidden files via:
terminal command:
defaults write com.apple.finder AppleShowAllFiles TRUE;killall Finder

closed 2019-01-06 by MacFly77

MacFly77, 2019-01-07:
Made small Automator scripts to help saving and updating the config file via iCloud.