bancika/diy-layout-creator #395: Will not run on Mac

gaznesh, 2019-01-08:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates

gaznesh, 2019-01-08:
Hi there. This software doesn't seem to run on my macbook I'm getting this error code:
[image]

bancika, 2019-01-08:
the solution is explained on a few places, one of which is literally on this page under **Before you proceed, please read the following:**

closed 2019-01-08 by bancika

gaznesh, 2019-01-08:
Well there's no need to be a twat about it. I can't see this error code where you say it is but fuck it, I'll go elsewhere for some software that works.

bancika, 2019-01-08:
Sorry if it came across as harsh. I get questions like this every day from people that cannot be bothered to to read the first paragraph of the manual that explains what needs to be done. Before you submit an issue report, it says

"If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491."

it's in the text that you submitted as the issue report, just scroll up.

It's not the app that's broken, it's Apple's bullshit policy. Unless I pay good money to get my app in the official store, they say that it's "broken". It's a totally free app, I cannot sink money into it in addition to tons of my time to develop and support it.

Cheers

gaznesh, 2019-01-08:
Hi there,
I must apologise too. I was too hasty to respond. I've got the software working and it's very good. The solution in my case was here: https://discussions.apple.com/thread/7682929?answerId=30706173022#30706173022

Thank you for writing this program,

All the best

Gareth Whittock

bancika, 2019-01-08:
Glad you like it, Gareth. All the best!