bancika/diy-layout-creator #396: DIYLC as a  Linux Snap

ChrisOfBristol, 2019-01-08:
The process for submitting an application to Debian in order to make DIYLC available on Linux is arcane and bureacratic, and GetDeb an alternative which hosted user-submitted applications appears to have retired. There is now a better way of distributing software called a Snap. I have created one for DIYLC, but have got stuck as I don't know Java, I think it's a configuration file access problem. Have error log - could you have a look?

bancika, 2019-01-08:
Sure. Please attach it here. I saw your email but didn't get to respond yet, as I was on vacation. It sounds interesting. I do have Ubuntu on VM that I can use to test. What's the procedure for packaging a snap? Maybe I can automate it as part of my usual build process that outputs windows installer, OSX app package and generic zip app.

ChrisOfBristol, 2019-01-08:
The Snap file type is not permitted - any ideas? See me email with some more information.

bancika, 2019-01-08:
I meant the log, you should be able to attach that

ChrisOfBristol, 2019-01-08:
Not a log, just error output.
chris@desktop:~$ myapp
19:52:24,540 DEBUG [main] myappStarter - Java version: 1.8.0_191-8u191-b12-0ubuntu0.16.04.1-b12 by Oracle Corporation
19:52:24,541 DEBUG [main] myappStarter - OS: Linux 4.15.0-43-generic
19:52:24,541  INFO [main] myappStarter - Starting myapp with working directory /snap/myapp/x12

(process:25267): Gtk-WARNING **: Locale not supported by C library.
	Using the fallback 'C' locale.
Gtk-Message: Failed to load module "pantheon-filechooser-module"
Gtk-Message: Failed to load module "canberra-gtk-module"
19:52:25,937 ERROR [main] myappStarter - Could not set Look&Feel
java.lang.NullPointerException
	at sun.awt.FontConfiguration.getVersion(FontConfiguration.java:1264)
	at sun.awt.FontConfiguration.readFontConfigFile(FontConfiguration.java:219)
	at sun.awt.FontConfiguration.init(FontConfiguration.java:107)
	at sun.awt.X11FontManager.createFontConfiguration(X11FontManager.java:774)
	at sun.font.SunFontManager$2.run(SunFontManager.java:431)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.font.SunFontManager.<init>(SunFontManager.java:376)
	at sun.awt.FcFontManager.<init>(FcFontManager.java:35)
	at sun.awt.X11FontManager.<init>(X11FontManager.java:57)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at java.lang.Class.newInstance(Class.java:442)
	at sun.font.FontManagerFactory$1.run(FontManagerFactory.java:83)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.font.FontManagerFactory.getInstance(FontManagerFactory.java:74)
	at java.awt.Font.getFont2D(Font.java:491)
	at java.awt.Font.access$000(Font.java:224)
	at java.awt.Font$FontAccessImpl.getFont2D(Font.java:228)
	at sun.font.FontUtilities.getFont2D(FontUtilities.java:180)
	at sun.font.FontUtilities.getCompositeFontUIResource(FontUtilities.java:399)
	at com.sun.java.swing.plaf.gtk.PangoFonts.lookupFont(PangoFonts.java:204)
	at com.sun.java.swing.plaf.gtk.GTKLookAndFeel.getGTKStyleFactory(GTKLookAndFeel.java:1603)
	at com.sun.java.swing.plaf.gtk.GTKLookAndFeel.loadStyles(GTKLookAndFeel.java:1567)
	at com.sun.java.swing.plaf.gtk.GTKLookAndFeel.initialize(GTKLookAndFeel.java:1452)
	at javax.swing.UIManager.setLookAndFeel(UIManager.java:538)
	at javax.swing.UIManager.setLookAndFeel(UIManager.java:583)
	at org.myapp.myappStarter.main(Unknown Source)
19:52:25,997 DEBUG [Thread-1] myappStarter - Starting font pre-loading
Exception in thread "Thread-1" java.lang.ExceptionInInitializerError
	at org.myapp.myappStarter$2.run(Unknown Source)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at sun.awt.FontConfiguration.getVersion(FontConfiguration.java:1264)
	at sun.awt.FontConfiguration.readFontConfigFile(FontConfiguration.java:219)
	at sun.awt.FontConfiguration.init(FontConfiguration.java:107)
	at sun.awt.X11FontManager.createFontConfiguration(X11FontManager.java:774)
	at sun.font.SunFontManager$2.run(SunFontManager.java:431)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.font.SunFontManager.<init>(SunFontManager.java:376)
	at sun.awt.FcFontManager.<init>(FcFontManager.java:35)
	at sun.awt.X11FontManager.<init>(X11FontManager.java:57)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at java.lang.Class.newInstance(Class.java:442)
	at sun.font.FontManagerFactory$1.run(FontManagerFactory.java:83)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.font.FontManagerFactory.getInstance(FontManagerFactory.java:74)
	at sun.java2d.SunGraphicsEnvironment.getFontManagerForSGE(SunGraphicsEnvironment.java:190)
	at sun.java2d.SunGraphicsEnvironment.getAvailableFontFamilyNames(SunGraphicsEnvironment.java:224)
	at sun.java2d.SunGraphicsEnvironment.getAvailableFontFamilyNames(SunGraphicsEnvironment.java:252)
	at org.myapp.swingframework.FontChooserComboBox.<clinit>(FontChooserComboBox.java:64)
	... 2 more
Exception in thread "main" java.lang.NullPointerException
	at javax.swing.MultiUIDefaults.getUIError(MultiUIDefaults.java:130)
	at javax.swing.UIDefaults.getUI(UIDefaults.java:761)
	at javax.swing.UIManager.getUI(UIManager.java:1016)
	at javax.swing.JPanel.updateUI(JPanel.java:126)
	at javax.swing.JPanel.<init>(JPanel.java:86)
	at javax.swing.JPanel.<init>(JPanel.java:109)
	at javax.swing.JPanel.<init>(JPanel.java:117)
	at javax.swing.JRootPane.createGlassPane(JRootPane.java:546)
	at javax.swing.JRootPane.<init>(JRootPane.java:366)
	at javax.swing.JFrame.createRootPane(JFrame.java:286)
	at javax.swing.JFrame.frameInit(JFrame.java:267)
	at javax.swing.JFrame.<init>(JFrame.java:234)
	at org.myapp.swing.gui.MainFrame.<init>(Unknown Source)
	at org.myapp.myappStarter.main(Unknown Source)

kroma-1, 2019-01-18:
This would be great. I offer myself for testing.

ChrisOfBristol, 2019-01-18:
I've added some packages as 'swing' is mentioned, but it still fails. I don't know any Java and am only just starting to learn Snapcraft, so I'm not really getting anywhere. Have you any knowledge of either, and if so, would you be willing to try building the Snap? I would send you all the relevant files.

ChrisOfBristol, 2019-01-19:
@kroma-1 If you just want to run DIYLC on your Linux computer, I can let you have a .deb for it.

kroma-1, 2019-01-20:
@ChrisOfBristol Unfortunately i know nothing about programming. I only meant I`m willing to test stability, usage and the like.
Please do send me that .deb file.
Thank you.

ChrisOfBristol, 2019-01-20:
@kroma-1 It's not permitted to attach it here, but if you contact me using the email address of my username followed by 'at' and ymail 'dot' com, I can email it to you, it's not the latest version, but works fine on my Ubuntu-Budgie desktop. I might update it if I can find the software and remember how :) It's well worth trying as it is excellent software, it does what you need without any flash rubbish.

ChrisOfBristol, 2019-01-22:
Another thought on Snaps. The DIYLC zip file is byte code (I think :). It may be that it needs to be source code to get into the Snap store.

ChrisOfBristol, 2019-01-22:
Another thought on Snap creation. I think the DIYLC zip file contains byte-code, and getting a Snap into the Snap store probably requires source-code.

ChrisOfBristol, 2019-01-27:
@kroma-1 Deb now available for latest version V3.50.

ChrisOfBristol, 2019-03-20:
Change of tack! After quite a lot of work on Snaps I've realised that they are a) too complex for me especially with my lack of knowledge of Java b) proprietary - owned by Canonical (Ubuntu). A 'Flatpak' would be much better. It looks as though help to do it is available too. Will start another thread if there is any progress on it.

closed 2019-03-20 by ChrisOfBristol