bancika/diy-layout-creator #397: Project's default font is not respected or editable since 3.47

gtranter, 2019-01-10:
In 3.47, the new default font Square721 BT is automatically applied to existing projects! This is very undesirable!

To reproduce, open any project that uses some other default font, make any change and then perform an undo (ctrl-z or menu). The font change takes place when the undo is performed. If the project is saved, the default font for the project is changed in the .diylc file.

It doesn't seem to help to change it back either. The next undo will reset it.

So it becomes impossible to use any font except Square721 BT (unless you are so good that you never have to undo).

bancika, 2019-01-10:
Thanks for reporting. Hmm, this seems to be an old bug. I would bet that if you had a project with a non-Tahoma font, it would revert back to Tahoma the same way. I'll add it to the pipeline.

bancika, 2019-01-10:
fixed for 3.48

closed 2019-01-10 by bancika