bancika/diy-layout-creator #401: Am I missing something?  Using Ant 1.7.0, Java jre1.8.0_191 and MyEclipse 2017 I get a build error when running build.xml for diylc-swing.  Error is:

shouse1953, 2019-01-13:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2019-01-13:
you need to add jarbundler JAR to ant classpath. The jar itself is located in ant-lib folder.
That build.xml builds a deliverable for mac together with windows installer and linux zip.

closed 2019-01-13 by bancika

shouse1953, 2019-01-13:
Thx.  Odd though as it had been building before.

shouse1953, 2019-01-13:
Thanks you, I added it to the ant classpath in MyEclipse and get beyond that error now.  However I now get this error:

BUILD FAILED

D:\code\diylc\src\diylc\diylc-swing\build.xml:119: Execute failed: java.io.IOException: Cannot run program "innosetup\iscc.exe" (in directory "D:\code\diylc\src\diylc\diylc-swing"): CreateProcess error=2, The system cannot find the file specified

The ISCC.exe program DOES exist in the diylc-swing\innosetup folder.

Is there another MyEclipse setting that needs to be set?  

 

Thx!

 

From: Steve House <steve@stevehouse.com> 
Sent: Sunday, January 13, 2019 3:10 PM
To: bancika/diy-layout-creator <reply@reply.github.com>
Subject: Re: [bancika/diy-layout-creator] Am I missing something? Using Ant 1.7.0, Java jre1.8.0_191 and MyEclipse 2017 I get a build error when running build.xml for diylc-swing. Error is: BUILD FAILED D:\code\diylc\src\diylc\diylc-swing\build.xml:96: taskdef c

 

Thx.  Odd though as it had been building before.