bancika/diy-layout-creator #402: sith WIndows 10, MyEclipse 2017, java 1.8 & ant 1.7, I get this error when I run execute ant build on build.xml:

shouse1953, 2019-01-14:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

shouse1953, 2019-01-14:
MyEclipse issue.

closed 2019-01-14 by shouse1953

bancika, 2019-01-14:
Cool, btw, can you please follow the convention of shorter issue title and put all the info in the description. It's easier to follow.

shouse1953, 2019-01-14:
Yep