bancika/diy-layout-creator #406: Dual monitor Display issue in 49.0

EL34xyz, 2019-01-16:
I just DL'd and installed 49.0. Found a weird bug. 
I run two 23 inch LED monitors in windows 7 pro.
Running the program on the main monitor is fine.
Dragging the program to the extended monitor results in the work area going black. 
The tools on the left side are fine, just the work area goes black

Here's a screen shot of that and my windows monitor setup

[image]

[image]

EL34xyz, 2019-01-16:
Bancika told me to comment out a couple lines in this file
It works fine now
C:\Program Files (x86)\DIYLC\config.properties

[image]

bancika, 2019-01-17:
fixed for 3.50

closed 2019-01-17 by bancika